openbmc/openbmc #3091: Enable CI on phosphor-webui

gtmills, 2018-04-13:
We need to do CI testing on phosphor-webui. 
"npm install" to begin with.

gtmills, 2018-04-13:
Would be nice to get a style formatter going for that repo as well.

edtanous, 2018-05-02:
clang-format is capable of formatting javascript.  We just need to define a clang-format file, and update the jenkins scripts.

As far as testing "npm install" isn't that covered by the bitbake recipe?

gtmills, 2018-05-02:
For non-openbmc/openbmc repos, CI just compiles the repo (assuming it is a c++ repo) and runs unit tests.
Only openbmc/openbmc commits use bitbake. For openbmc/openbmc commits, CI builds several images and does limited QEMU and hardware testing (https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/). 
When we bump the Web UI recipe (e.g. https://gerrit.openbmc-project.xyz/#/c/10388/) we would fail if "npm install" fails but it would nice to test "npm install" on the individual commits, a lot of times the bumps are several commits and if the bump fails, that code is already merged into Web UI.. Would be nice to catch it before it is merged. I think the styling and unit testing are more important than "npm install" but still think "npm install" would be nice to have.

edtanous, 2018-05-02:
Gotcha.  I think all of that makes sense to do.

geissonator, 2018-05-25:
ok, we've got the basic clang-format validation in now for the repo. 

Next we need to figure out the unit testing aspect.  There's a good right up at https://docs.angularjs.org/guide/unit-testing.  Be good to start out with something basic that builds the app and verifies the basic login/status page.

gtmills, 2018-09-25:
Had a start here: https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc-build-scripts/+/10004/ but never got back to it

closed 2019-04-26 by stale[bot]