openbmc/openbmc #3092: [Witherspoon]: Stop at uboot after reboot the BMC

ouyangxx10, 2018-04-13:
After reboot the BMC, the BMC stop at uboot. Discovery of environment variable loss.
```
U-Boot 2016.07 (Apr 08 2018 - 21:04:47 -0400)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   MAC0 : RMII/NCSI
MAC1 : RGMII
FTGMAC100#0
Error: FTGMAC100#0 address not set.
, FTGMAC100#1
Error: FTGMAC100#1 address not set.

Hit any key to stop autoboot:  0
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast#
ast# env print
77a2fa21=0
baudrate=115200
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
bootcmd=bootm 20080000
bootdelay=2
ethact=FTGMAC100#0
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
verify=yes

Environment size: 217/65531 bytes
ast#
```

amboar, 2018-04-13:
Can you provied the ouput of `ubi info l`? It will be telling if there are any UBI volumes on the flash.

You mention reboot, so I assume you don't mean PDU cycled. If the system was not PDU-cycled, it's worth providing the output of `md.l 0x1e785030 1` at the u-boot prompt. This will tell us if your system has switched its BMC flash chips over the reboot.

amboar, 2018-04-13:
@ouyangxx10 as a follow up, we need a bit more information. Can you please link to the OpenBMC commit you built the image from that was on the flash when the failure occurred? And if you have any other information about what you were doing before the reboot that would be great, logs especially.

ouyangxx10, 2018-04-13:
@amboar  I will make detail description:
My build env is:
```
Build Configuration:
BB_VERSION = "1.36.0"
BUILD_SYS = "x86_64-linux"
NATIVELSBSTRING = "ubuntu-16.04"
TARGET_SYS = "arm-openbmc-linux-gnueabi"
MACHINE = "witherspoon"
DISTRO = "openbmc-phosphor"
DISTRO_VERSION = "0.1.0"
TUNE_FEATURES = "arm armv6 thumb arm1176jzs"
TARGET_FPU = "soft"
meta
meta-poky
meta-oe
meta-networking
meta-perl
meta-python
meta-virtualization
meta-phosphor
meta-aspeed
meta-ast2500
meta-openpower
meta-ibm
meta-witherspoon
workspace = "master:65b556c79c4af44c69f3fb278e63515a5346a3bd"
```

when I flash the image to board, can't enter into BMC file system, please refer to #3070 .
I made a little change of the source code(kernel). I found the driver cannot get the "JEDEC id", so I write a temporary JEDEC id in the driver, then the bmc can enter into file system, reboot it, it will stop at uboot.
```
ast# ubi info 1
Error, no UBI device/partition selected!
ast# ubi info l
Error, no UBI device/partition selected!
ast#
ast# md.l 0x1e785030 1
1e785030: 00000100                               ....
ast#
```

ouyangxx10, 2018-04-13:
@amboar  first time boot log same as #3070 .
the detail reboot log as follow(confirm not do any opearate, just login and reboot):
```
[  OK  ] Reached target Login Prompts.
[  OK  ] Started Phosphor Dump Manager.
[  OK  ] Started Network Name Resolution.
[  OK  ] Reached target Host and Network Name Lookups.
[FAILED] Failed to start Remount the ubi volume after a reboot.
See 'systemctl status obmc-flash-bmc-ubiremount.service' for details.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: root
Password:
root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~# reboot
         Stopping Lightweight SLP Server...
         Stopping Phosphor Inband IPMI...
         Stopping Phosphor Log Manager...
         Stopping Phosphor DBus Monitor...
[  OK  ] Removed slice system-org.openbmc.buttons.reset.slice.
         Stopping Phosphor Dump core monitor....
         Stopping Phosphor Console Muxer listening on device /dev/ttyVUART0...
         Stopping FRU Fault monitor service...
[  OK  ] Stopped Wait for /xyz/openbmc_project/time/owner.
[  OK  ] Stopped Wait for /xyz/openbmc_project/inventory/system/chassis/motherboard/powersupply0.
[  OK  ] Removed slice system-obmc\x2dled\x2dgroup\x2dstart.slice.
[  OK  ] Removed slice system-org.openbmc.control.Power.slice.
         Stopping Phosphor Dump Manager...
[  OK  ] Stopped Wait for /xyz/openbmc_project/time/sync_method.
         Unmounting /usr/local...
[  OK  ] Stopped target Timers.
[  OK  ] Removed slice system-org.openbmc.buttons.Power.slice.
[  OK  ] Stopped target Containers.
[  OK  ] Removed slice system-phosphor\x2dreset\x2dsensor\x2dstates.slice.
         Stopping Phosphor Download Manager...
         Stopping D-Bus System Message Bus...
[  OK  ] Stopped Wait for /xyz/openbmc_project/control/host0/auto_reboot.
[  OK  ] Removed slice system-phosphor\x2ddiscover\x2dsystem\x2dstate.slice.
         Stopping Phosphor Settings Daemon...
         Stopping Phosphor GPIO 0 Presence...
         Stopping Phosphor GPIO 1 Presence...
         Stopping System Logging Service...
[  OK  ] Removed slice system-phosphor\x2dcooling\x2dtype.slice.
         Stopping Phosphor IPMI BT DBus Bridge...
[  OK  ] Removed slice system-witherspoon\x2dpower\x2dsupply\x2dmonitor.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/inventory/system/chassis/motherboard/powersupply1.
         Stopping Phosphor User Manager...
[  OK  ] Stopped Wait for /xyz/openbmc_project/led/groups/bmc_booted.
[  OK  ] Stopped target Login Prompts.
         Stopping Serial Getty on ttyS4...
[  OK  ] Stopped Avahi mDNS/DNS-SD Stack.
[  OK  ] Stopped Phosphor Settings Daemon.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped Phosphor Console Muxer listening on device /dev/ttyVUART0.
[  OK  ] Stopped Phosphor Dump core monitor..
[  OK  ] Stopped Phosphor Download Manager.
[  OK  ] Stopped Phosphor Log Manager.
[  OK  ] Stopped Phosphor User Manager.
[  OK  ] Stopped Phosphor IPMI BT DBus Bridge.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped SSH Key Generation.
[  OK  ] Stopped Phosphor Dump Manager.
[  OK  ] Stopped Lightweight SLP Server.
[  OK  ] Stopped Network Name Resolution.
[  OK  ] Stopped Serial Getty on ttyS4.
[... 109 lines cut ...]
[   66.036030] systemd-shutdown[1]: Unmounting /media.
[   66.050115] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[   66.212333] UBIFS (ubi0:2): un-mount UBI device 0
[   66.258741] reboot: Restarting system

U-Boot 2016.07 (Apr 08 2018 - 21:04:47 -0400)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   MAC0 : RMII/NCSI
MAC1 : RGMII
FTGMAC100#0
Error: FTGMAC100#0 address not set.
, FTGMAC100#1
Error: FTGMAC100#1 address not set.

Hit any key to stop autoboot:  0
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast#
```

anoo1, 2018-04-13:
For the last week, master had a change that causes the BMC chip to switch to the alternate one on reboot. That's probably what is being seen here where the alt chip has an old u-boot env variable or has not been programmed, so when the system switched to the alt chip it appears as if it was the BMC chip that lost its information. Issue https://github.com/openbmc/openbmc/issues/3094

closed 2018-04-17 by ouyangxx10