openbmc/openbmc #3096: OP920: FVT : Software DeleteAll image - ubirmvol: error!: cannot UBI remove volume

cdeadmin, 2018-04-16:


anoo1, 2018-04-16:
Test: Install multiple BMC or PNOR images and DeleteAll images interface
```
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST  -d '{"data": []}' https://$BMC_IP/xyz/openbmc_project/software/action/DeleteAll                         {
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```
```
root@witherspoon:~# journalctl -f
-- Logs begin at Tue 2018-03-06 18:41:24 UTC. --
Mar 06 18:41:46 witherspoon kernel: block ubiblock0_1: released
Mar 06 18:41:46 witherspoon kernel[1039]: [  324.060000] block ubiblock0_1: released
Mar 06 18:41:47 witherspoon kernel: ubi4 error: ubi_open_volume: cannot open device 4, volume 2, error -16
Mar 06 18:41:47 witherspoon obmc-flash-bmc[1465]: ubirmvol: error!: cannot UBI remove volume
Mar 06 18:41:47 witherspoon obmc-flash-bmc[1465]:           error 16 (Device or resource busy)
Mar 06 18:41:47 witherspoon kernel[1039]: [  324.450000] ubi4 error: ubi_open_volume: cannot open device 4, volume 2, error -16
Mar 06 18:41:47 witherspoon systemd[1]: obmc-flash-bmc-ubiro-remove@b0588197.service: Main process exited, code=exited, status=255/n/a
Mar 06 18:41:47 witherspoon systemd[1]: Failed to start Deletes read-only and kernel ubi volume b0588197.
Mar 06 18:41:47 witherspoon systemd[1]: obmc-flash-bmc-ubiro-remove@b0588197.service: Unit entered failed state.
Mar 06 18:41:47 witherspoon systemd[1]: obmc-flash-bmc-ubiro-remove@b0588197.service: Failed with result 'exit-code'.
Mar 06 18:41:52 witherspoon systemd[1]: Started Updates the u-boot variable to point BMC version to 84fb16be.
Mar 06 18:41:57 witherspoon systemd[1]: Started Clean up UBI volumes unattached to a BMC version.
```

The images are deleted but seen the above error on the journald.

anoo1, 2019-02-20:
I believe this has been fixed.

closed 2019-02-20 by anoo1