openbmc/openbmc #3098: openpower-occ-control: Bind to OCC asynchronously on active sensor set

geissonator, 2018-04-16:
Binding to the OCC when the host sends the set sensor command for the OCC can take a fairly lengthy period of time for OpenBMC.  After some discussion, it was decided it would be best to have openpower-occ-control make this call asynchronous (i.e. host ipmid routes the set sensor command over to openpower-occ-control which starts the process of binding but returns to host ipmi right after that).

Ideally we do something with the sd_event loop and avoid a separate thread. Design needs some investigation.

Maybe something in the bind interface in https://github.com/openbmc/openpower-occ-control/blob/master/occ_device.hpp

amboar, 2018-04-17:
I don't know how userspace will cope, but I've seen *really* bad things happen in the kernel when requesting 0 second timers. That was the primary cause of the OCC performance issues we had earlier in the year.

dkodihal, 2018-04-17:
sdbusplus has an async D-Bus method call API. Last time I used that in host-ipmi to call openpower-occ-control, it didn't really help. While the set-sensor for OCC was much faster this way, it didn't really help with performance - whenever the bind occurred there was a slowdown with the bmc performance.

dkodihal, 2018-04-17:
I think we'd have to look up why binding the OCC driver locks up the CPU. If that's something that's fixed in the 4.13 kernel, we shouldn't be needing to do this anyway?

mine260309, 2018-04-17:
@dkodihal Yup I have tested OCC on dev-4.13, all the issues I met on 4.10 are fixed, no bind failure, no slow down.

@amboar The 0 second timer I mentioned above is in user space with sd_event loop, basically it just defers a function call by putting it in event loop. It does no harm and is not related to kernel.

amboar, 2018-04-17:
@mine260309 great, thanks for clearing that up

dkodihal, 2018-04-17:
@mine260309 ok, thanks. Then I guess host-ipmi can make an async d-bus call to occ-control, although host-ipmi has mostly common code in terms of routing set sensors, so we'd need to figure out how to do this without breaking the generic code.

geissonator, 2018-04-17:
@dkodihal - the hope was to put this asynch aspect in openpower-occ-control so host-ipmid generic code doesn't have to change it's behavior.

ojayanth, 2018-04-24:
in the latest build is taking , 1 sec for OCC0 and less than 3 seconds( some time 2 seconds) for OCC1.
ipmi_sen_set_sensor: (Start) Tue Apr 24 12:22:52 2018
 	 SENSOR_NUM=8 	ipmi_sen_set_sensor(End)  Tue Apr 24 12:22:53 2018

ipmi_sen_set_sensor: (Start) Tue Apr 24 12:22:53 2018
 	 SENSOR_NUM=9 	ipmi_sen_set_sensor(End)  Tue Apr 24 12:22:56 2018

dkodihal, 2018-04-24:
3 seconds is a concern (can easily hit the 5s timeout). The time you've noted here is within the host ipmid process. There's also the overhead of the host and BMC BT driver, the IPMI BT userspace application on the BMC, and the D-Bus communication between that app and host-ipmid. So this sounds like we do need to improve on the time.

ojayanth, 2018-04-25:
@dkodihal No much overhead in Host and BMC BT driver.  Below data collected from issuing IPMI command from Host.
 # date;ipmitool raw 0x04 0x30 0x08 0xa8 0x00 0x02 0x00 0x01 0x00 0x00 0x00 0x0
0;date;ipmitool raw 0x04 0x30 0x09 0xa8 0x00 0x02 0x00 0x01 0x00 0x00 0x00 0x00;
date
Wed Apr 25 13:52:54 UTC 2018

Wed Apr 25 13:52:55 UTC 2018

Wed Apr 25 13:52:57 UTC 2018

amboar, 2018-04-29:
@ojayanth FWIW, if you're trying to time the execution of applications, try the [`time`](https://ss64.com/bash/time.html) bash utility, or /usr/bin/time

amboar, 2018-04-29:
@ojayanth also, please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

amboar, 2019-04-02:
Binding the OCC is still fairly performance-intensive; it's not hard to trigger hostboot crashes due to poor IPMI performance around this point. I think the issue should remain open.

closed 2019-11-01 by stale[bot]