openbmc/openbmc #3102: Need to update the "Server power" in the header and "Current status" bar in the power operations

gtmills, 2018-04-17:
Need to update the "Server power" in the header and "Current status" bar on the power operations page with the current power status. 
https://github.com/openbmc/openbmc/issues/2758 is a related story, about updating the  "Current status" bar on the power operations page during a boot. 

Could look at using an Event subscription
 https://github.com/openbmc/docs/blob/master/rest-api.md#event-subscription-protocol

jaypadath, 2018-04-26:
I am able to use the websockets to receive the notification of current host change. Patch is working good while trying the power on/off from witherspoon commanline.

But the behaviour of power on/off from GUI is not affected with change. Need see whether #2758 addresses this scenario.

jaypadath, 2018-04-26:
With a new patch, I am able to make the behaviour consistent with GUI and command line power on/off operations.

rfrandse, 2018-04-27:
<https://gerrit.openbmc-project.xyz/10338> Update the server power state dynamically in GUI
Resolves  #3102 Need to update the "Server power" in the header and "Current status" bar in the power operations

closed 2018-06-28 by geissonator