openbmc/openbmc #3104: OP920:Witherspoon:yob46cbmc:Server name edit does not save the new value

cdeadmin, 2018-04-17:


gtmills, 2018-04-17:
From the overview page
Click on the edit link next to the Server name (called Server name but it is actually the hostname from  `/xyz/openbmc_project/network/config/attr/HostName`)
[image]

[image]

When you enter a new name and click, nothing happens. 
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://$bmc/xyz/openbmc_project/network/config/attr/HostName
{
  "data": "witherspoon"

```

gtmills, 2018-04-17:
Commented this edit button out for now, https://gerrit.openbmc-project.xyz/#/c/10197/1

gtmills, 2018-04-28:
same issue here https://github.com/openbmc/openbmc/issues/2730

rfrandse, 2018-05-04:
<https://gerrit.openbmc-project.xyz/10442> Allow user to modify hostname in overview page
Resolves  #3104 Host name edit does not save the new value

closed 2018-05-14 by geissonator