openbmc/openbmc #3107: MPV:OPRAS920:Witherspoon:Loss of A/C on PSU or Removal of PSU results in SEL : Unknown #0x0c sensor assert

cdeadmin, 2018-04-18:


amboar, 2018-04-23:
What is OPRAS920?

gtmills, 2019-01-23:
I assume it was supposed to be a internal IBM release. :(

gtmills, 2019-01-23:
@tomjoseph83 Still a problem? Can you add some detail?

bjwyman, 2019-07-25:
This caught my attention in the #openbmc-bot IRC channel, as it had A/C and PSU in the title. I took a peak at the internal defect.

This had something to do with a SEL elist having a column set to "Unknown #0x0c" when there was a power supply input fault (Loss of A/C). The SEL elist was retrieved via IPMI, looking for a "fault sensor" to show up. 

"Witherspoon has no sensor number associated for the power supply fault, IPMI sensor numbers are exhausted for Witherspoon. So IPMI SEL record will show the called out inventory as motherboard."

"The IPMI sensor numbers are limited to 255."

Move to using Redfish to show specific power faults....

gtmills, 2019-07-25:
@bjwyman asked me to close this.

closed 2019-07-25 by gtmills