openbmc/openbmc #3108: [LTCTest][OPAL][OP910] IPMITOOL Set watchdog command makes the Chassis and HOST to Reboot

cdeadmin, 2018-04-18:


dkodihal, 2018-04-23:
This can be reproduced on 
https://github.com/openbmc/openbmc/commit/abf205c922186290ced31166664c3e54c62b64ec.

Via inband IPMI, set the host watchdog :
`ipmitool -I open raw 0x06 0x24 0x44 0x00 0x00 0x10 0xc8 0x00`

There's no watchdog registered in the SEL entries.

dkodihal, 2018-05-18:
This is working as designed.

Corrected the headline to describe accurately the issue raised by the submitter.

On the expiry of the watchdog timer, the corresponding systemd unit is invoked to reboot the host.

There is no error log whenever the host is rebooted, because of which there is no entry in the SEL to correspond to that.

The SEL is currently a mapping of the error logs in the bmc and all the error logs are mapped to System event(System Event #0x01 | Undetermined system hardware failure | Asserted), the only other case is when there is a FRU callout.(Processor #0xa4 | Disabled | Asserted).

closed 2018-05-18 by dkodihal