openbmc/openbmc #3112: SW425373 : STC:OP920:Witherspoon:Non polurated GPSs sonsor data is displayed on GUI

cdeadmin, 2018-04-18:


spinler, 2018-04-18:
On the IBM witherspoon system, we sell a 2 GPU config, but the OCC device driver is always giving us power values for 6 GPUs, so the extra ones still get put onto D-Bus and make their way into the web UI.

It would be nice for us to not make D-Bus objects for sensors on missing FRUs.  On GPUs it is complicated a bit in that hostboot always tells us all 4/6 are present when they boot, and then later when the OCC starts they come back and tell us which ones are actually there.

spinler, 2018-04-18:
This would also be useful for removing the fan 1 sensors on water cooled witherspoons that don't use that fan.  In this case we could check the water-cooled D-Bus property.

spinler, 2018-04-25:
Decided not to push this complexity into phosphor-hwmon.  Will be handled by having something choose which config file phosphor-hwmon should use before it starts.

closed 2018-04-25 by spinler