openbmc/openbmc #3115: avahi: bin-utils adds 3MB to the image size

anoo1, 2018-04-19:
Commit https://github.com/openbmc/openbmc/commit/8f852d2b9484202118e382a31d16e659b91789ee added bin-utils to the avahi recipe, which is increasing the image size by 3MB.

The only needed command from bin-utils is "host" so this issue is to remove everything that is not necessary out of the package since 3MB is a very big increase in a limited flash size environment like the BMC.

gtmills, 2018-08-20:
@anoo1 Did avahi get backed out? Don't see it under https://github.com/openbmc/openbmc/tree/master/meta-phosphor/common/recipes-connectivity

anoo1, 2018-08-27:
Right, it was not merged because of the size issue. @ratagupt , are we still trying to fit avahi in the bmc or we've moved to other solutions?

closed 2019-04-26 by stale[bot]