openbmc/openbmc #3117: Need a way to get the interfaces via REST.

gtmills, 2018-04-19:
We had talked about a virtual 'interfaces' property. Ed also mentioned a rest API that exposes Interfaces.
Need to know which objects have the system interface, https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Inventory/Item/System.interface.yaml
This came from discussion on https://gerrit.openbmc-project.xyz/#/c/10124.

edtanous, 2018-05-01:
I think the easiest way to explain would be with a simple example.  The existing API looks similar to this:

GET https://10.243.49.13:18080/xyz/openbmc_project/software/enumerate

{
    "data": {
        "/xyz/openbmc_project/software/34925e9a": {
            "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
            "Path": "",
            "Priority": 0,
            "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
            "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
            "Version": "v2.1-302-g4c12c17",
            "associations": ""
        }
    },
    "message": "200 OK",
    "status": "ok"
}

This return is becoming problematic, because in the case of inventory (and other things) we need to know what interfaces the properties exist on, to avoid collisions in property names, handle interfaces that might not have any properties defined (like System) and make implementation specific switches based on whether or not the interface exists.  Something that I would propose would be something like this, and would more closely match what the mapper returns

GET https://10.243.49.13:18080/xyz/openbmc_project/software/better_enumerate

{
    "data": {
        "/xyz/openbmc_project/software/34925e9a": {
            "xyz.openbmc_project.Software.Activation": {
            "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
            "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
          },
            "xyz.openbmc_project.Software.RedundancyPriority": {
            "Priority": 0,
          },
            "xyz.openbmc_project.Common.FilePath": {
            "Path": ""
          },
            "xyz.openbmc_project.Software.Version": {
            "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
            "Version": "v2.1-302-g4c12c17",
          },
            "org.openbmc.Associations": {
            "associations": ""
          }
        }
    },
    "message": "200 OK",
    "status": "ok"
}

The key difference here is that this one enumerates the interfaces, as well as the properties, which gives you as a front end developer a nicer interface to code against.

geissonator, 2018-05-01:
ok, makes sense, but I'm still not seeing a direct solution with my problem in https://gerrit.openbmc-project.xyz/#/c/10310/.  In that situation we're doing an enumerate on the inventory, and we happen to have some non-inventory items in there like code activation information.  If I introspect them and look at their properties and interfaces, I still don't really see what the client could do to differentiate?  Here's the activation and then something we do want, the chassis.

```
# busctl introspect xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/inventory/system/chassis/activation | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE FLAGS
org.freedesktop.DBus.Introspectable interface -         -            -
.Introspect                         method    -         s            -
org.freedesktop.DBus.Properties     interface -         -            -
.Get                                method    ss        v            -
.GetAll                             method    s         a{sv}        -
.Set                                method    ssv       -            -
.PropertiesChanged                  signal    sa{sv}as  -            -
org.openbmc.Association             interface -         -            -
org.openbmc.Object.Properties       interface -         -            -
.SetMultiple                        method    sa{sv}    -            -
```

```
# busctl introspect xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory/system/chassis | cat
NAME                                                TYPE      SIGNATURE RESULT/VALUE FLAGS
org.freedesktop.DBus.Introspectable                 interface -         -            -
.Introspect                                         method    -         s            -
org.freedesktop.DBus.Peer                           interface -         -            -
.GetMachineId                                       method    -         s            -
.Ping                                               method    -         -            -
org.freedesktop.DBus.Properties                     interface -         -            -
.Get                                                method    ss        v            -
.GetAll                                             method    s         a{sv}        -
.Set                                                method    ssv       -            -
.PropertiesChanged                                  signal    sa{sv}as  -            -
xyz.openbmc_project.Inventory.Decorator.CoolingType interface -         -            -
.AirCooled                                          property  b         true         emits-change writable
.WaterCooled                                        property  b         false        emits-change writable
```

edtanous, 2018-05-01:
I don't have time to dig deeper today, but I would've expected that object in inventory manager to expose an xyz.openbmc_project.Inventory.Item.Chassis interface, with no properties attached https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Inventory/Item/Chassis.interface.yaml

In that case, when enumerating the new interface, you would do something like:
Do enumerate on chassis and put into chassis list
for chassis in chassis_list:
   if "xyz.openbmc_project.Inventory.Item.Chassis" in chassis.interface_list:
      // Do chassis specific things.

geissonator, 2018-05-02:
Hmm, ok, so the implication is the client, instead of just doing an inventory/enumerate, does a more targeted enumerate (inventory/system, inventory/chassis, inventory/power_supply, inventory/dimm, ...).  Or it does the full enumerate on inventory but has a way to inspect the better_enumerate data to figure out what it really wants to show?

edtanous, 2018-05-02:
The later.  My thinking was that the client would still do an inventory/enumerate, but in the data linked, you would be able to loop over the entries and find the ones that you cared about.  This is very similar to how the Thermal and chassis redfish schemas in bmcweb work, but it's doing it from the BMC side, so it has access to the interface information.
We could also give the ability to filter the API based on interface, but I don't really see the use case for it, as it's just as easy to loop in javascript/client side, and the extra entries don't really hurt you any.

Do you think you can link me a dump of the GetManagedObjects call that your system gives you for the inventory manager?  I suspect your system is more complete than the QEMU variant of witherspoon gives  me.
From there, I can doctor up an example return of what I'm thinking an inventory API in the rest server would look like.

bradbishop, 2018-05-02:
fwiw....the REST API (and therefore the DBus API) was originally viewed as a stable API and user-facing.  So we didn't want to expose the end-user to DBus things like interfaces.

Obviously the thinking has shifted to the DBus API is our internal API, so/and we don't want it to be stable.  And Redfish or IPMI will be/are the user-facing interfaces.  So exposing DBus concepts in the current API seems ok to me...long term the API is only meant for hacking anyway, right?

The problem I see is backward compatibility.  There are all kinds of tools out there consuming what was supposed to be a stable API but what is now an unstable API.  I'm not sure what to do about that.

geissonator, 2018-05-02:
Here's a dump from our witherspoon - https://gist.github.com/geissonator/055357a2bc308ac09d09df80133c8715

geissonator, 2018-05-07:
[system_managedobjects.txt](https://github.com/openbmc/openbmc/files/1980826/system_managedobjects.txt)

jinneal, 2018-09-20:
From QEMU of witherspoon:

why there are not Item.system or Item.Classis interface for the objs? 

root@witherspoon:~# dbus-send --system --print-reply --dest=xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory org.freedesktop.DBus.ObjectManager.GetManagedObjectsmethod return time=1537412210.647074 sender=:1.15 -> destination=:1.106 serial=54 reply_serial=2
   array [
      dict entry(
         object path "/xyz/openbmc_project/inventory/system"
         array [
            dict entry(
               string "org.freedesktop.DBus.Peer"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Properties"
               array [
               ]
            )
            dict entry(
               string "xyz.openbmc_project.Inventory.Decorator.AssetTag"
               array [
                  dict entry(
                     string "AssetTag"
                     variant                         string ""
                  )
               ]
            )
         ]
      )
      dict entry(
         object path "/xyz/openbmc_project/inventory/system/chassis"
         array [
            dict entry(
               string "org.freedesktop.DBus.Peer"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Properties"
               array [
               ]
            )
            dict entry(
               string "xyz.openbmc_project.Inventory.Decorator.CoolingType"
               array [
                  dict entry(
                     string "AirCooled"
                     variant                         boolean true
                  )
                  dict entry(
                     string "WaterCooled"
                     variant                         boolean false
                  )
               ]
            )
         ]
      )
      dict entry(
         object path "/xyz/openbmc_project/inventory/system/chassis/motherboard/powersupply1"
         array [
            dict entry(
               string "org.freedesktop.DBus.Peer"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Properties"
               array [
               ]
            )
            dict entry(
               string "xyz.openbmc_project.Inventory.Item"
               array [
                  dict entry(
                     string "PrettyName"
                     variant                         string "powersupply1"
                  )
                  dict entry(
                     string "Present"
                     variant                         boolean true
                  )
               ]
            )
         ]
      )
      dict entry(
         object path "/xyz/openbmc_project/inventory/system/chassis/motherboard/powersupply0"
         array [
            dict entry(
               string "org.freedesktop.DBus.Peer"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Properties"
               array [
               ]
            )
            dict entry(
               string "xyz.openbmc_project.Inventory.Item"
               array [
                  dict entry(
                     string "PrettyName"
                     variant                         string "powersupply0"
                  )
                  dict entry(
                     string "Present"
                     variant                         boolean true
                  )
               ]
            )
         ]
      )
   ]

closed 2019-04-26 by stale[bot]