openbmc/openbmc #3120: REST "Delete" logging failed

gkeishin, 2018-04-20:
```
$ ./rest_cmd -i xx.xx.xx.xx -o esel
Fetching esel from BMC
Error entry(s) found:
[u'/xyz/openbmc_project/logging/entry/6']
Entry : /xyz/openbmc_project/logging/entry/6
{
    "AdditionalData": [
        "_PID=1766",
        "STRING=FOO"
    ],
    "Description": "An internal BMC error occurred",
    "EventID": "None",
    "Id": 6,
    "Message": "example.xyz.openbmc_project.Example.Elog.AutoTestSimple",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1524235871471,
    "associations": []
}
```
`Delete` failed
```
$ ~/obmcutil-commands
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X DELETE https://$BMC_IP/xyz/openbmc_project/logging/entry/6
{
  "data": {
    "description": "The specified resource cannot be removed: '/xyz/openbmc_project/logging/entry/6'"
  },
  "message": "403 Forbidden",
  "status": "error"
}
```
`DeletAll` working though
```
$ ~/obmcutil-commands
$ curl -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data":[]}' https://$BMC_IP/xyz/openbmc_project/logging/action/DeleteAll
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
$ ~/obmcutil-commands
$ ./rest_cmd -i xx.xx.xx.xx -o esel
Fetching esel from BMC
No Error entry(s) found:

````

spinler, 2018-04-20:
I think I need to implement the Delete interface on the new com.ibm.Logging objects

gkeishin, 2018-04-20:
probably take a look at the dump too , just to be safe.

rfrandse, 2018-04-23:
<https://gerrit.openbmc-project.xyz/10275> Remove the interfaces removed handler
Resolves  #3120 REST "Delete" logging failed

closed 2018-04-25 by geissonator

gkeishin, 2018-04-26:
Tested on "v2.2-35-geee4542" latest build id #1397-eee4542ced3754 (Apr 25, 2018 8:14:11 PM)  looks like it's not yet into the build yet.