openbmc/openbmc #3124: [romulus]: BMC first time boot can not recognize the PHY chip

ouyangxx10, 2018-04-24:
When I flash the bmc image to flash chip, first time boot bmc, it can not recognize the PHY chip(Generic PHY), login in BMC debug console and reboot bmc, the second time boot it will recognize the PHY chip. At the same time, there will be log of kernel panic.
Debug console log:
```
......
[    2.973764] aspeed-smc 1e630000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[    2.981971] aspeed-smc 1e630000.flash-controller: Aspeed SMC probe failed -2
[    2.995042] aspeed-smc: probe of 1e630000.flash-controller failed with error -2
[    3.004462] libphy: Fixed MDIO Bus: probed
[    3.014321] ftgmac100 1e660000.ethernet: Generated random MAC address 56:40:8c:33:a6:eb
[    3.022341] ftgmac100 1e660000.ethernet: Using NCSI interface
[    3.029407] ftgmac100 1e660000.ethernet eth0: irq 19, mapped at 9f80a000
[    3.041446] ftgmac100 1e680000.ethernet: Generated random MAC address 4a:3b:a5:b0:a2:a5
[    3.058096] libphy: ftgmac100_mdio: probed
[    3.062330] Generic PHY 1e680000.ethernet--1:00: attached PHY driver [ Generic PHY ] (mii_bus:phy_addr=1e680000.ethernet--1:00, irq=-1)
[    3.085630] ftgmac100 1e680000.ethernet eth1: irq 20, mapped at 9f80c000
[    3.092627] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    3.099209] ehci-platform: EHCI generic platform driver
......
[  OK  ] Stopped Phosphor MBOX Daemon.
[FAILED] Failed to start Phosphor MBOX Daemon.
See 'systemctl status mboxd.service' for details.
[  OK  ] Started Phosphor Download Manager.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 romulus ttyS4

romulus login: [   60.109076] ------------[ cut here ]------------
[   60.113762] WARNING: CPU: 0 PID: 968 at /kernel-source//net/sched/sch_generic.c:316 dev_watchdog+0x22c/0x244
[   60.123590] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out
[   60.130382] CPU: 0 PID: 968 Comm: flasher.exe Not tainted 4.13.16-101790466fa90d70d475ef015df8aa1527a23a73 #5
[   60.140416] Hardware name: Generic DT based system
[   60.145275] [<80109930>] (unwind_backtrace) from [<80106bc4>] (show_stack+0x20/0x24)
[   60.153051] [<80106bc4>] (show_stack) from [<8057eba4>] (dump_stack+0x20/0x28)
[   60.160298] [<8057eba4>] (dump_stack) from [<80115ed4>] (__warn+0xe0/0x108)
[   60.167276] [<80115ed4>] (__warn) from [<80115f50>] (warn_slowpath_fmt+0x54/0x74)
[   60.174781] [<80115f50>] (warn_slowpath_fmt) from [<8049e1e0>] (dev_watchdog+0x22c/0x244)
[   60.183100] [<8049e1e0>] (dev_watchdog) from [<80158088>] (call_timer_fn+0x40/0x124)
[   60.190881] [<80158088>] (call_timer_fn) from [<80158220>] (expire_timers+0xb4/0xc0)
[   60.198645] [<80158220>] (expire_timers) from [<80158320>] (run_timer_softirq+0xa4/0x19c)
[   60.206835] [<80158320>] (run_timer_softirq) from [<80101600>] (__do_softirq+0xd0/0x2e4)
[   60.214940] [<80101600>] (__do_softirq) from [<80119f9c>] (irq_exit+0xf4/0x13c)
[   60.222277] [<80119f9c>] (irq_exit) from [<8014c8c0>] (__handle_domain_irq+0x60/0xb8)
[   60.230232] [<8014c8c0>] (__handle_domain_irq) from [<80101524>] (avic_handle_irq+0x68/0x70)
[   60.238707] [<80101524>] (avic_handle_irq) from [<80107d34>] (__irq_usr+0x54/0x80)
[   60.246276] Exception stack(0x9d175fb0 to 0x9d175ff8)
[   60.251332] 5fa0:                                     00f6d018 00000000 ffffffff 00f6d018
[   60.259507] 5fc0: 4def8c8c 4dedf94c 4def8884 4ddd9784 00000054 4e03c7cc 4e03c75c 00000005
[   60.267686] 5fe0: 4dd85194 7e8e8ae8 4ded2d98 4dd85190 20000010 ffffffff
[   60.274295] ---[ end trace fea232d330b7baa3 ]---
[   64.195313] random: crng init done

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 romulus ttyS4

romulus login:
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 romulus ttyS4

romulus login: root
Password:
root@romulus:~#
root@romulus:~#
root@romulus:~# reboot
[  OK  ] Stopped Start chassis0 on after BMC reset.
         Stopping Hostname Service...
[  OK  ] Stopped target Host0 running after reset.
         Stopping Phosphor sysfs LED controller...
......

[    2.973459] aspeed-smc 1e630000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[    2.981664] aspeed-smc 1e630000.flash-controller: Aspeed SMC probe failed -2
[    2.994719] aspeed-smc: probe of 1e630000.flash-controller failed with error -2
[    3.004127] libphy: Fixed MDIO Bus: probed
[    3.013973] ftgmac100 1e660000.ethernet: Read MAC address 56:40:8c:33:a6:eb from chip
[    3.021818] ftgmac100 1e660000.ethernet: Using NCSI interface
[    3.028889] ftgmac100 1e660000.ethernet eth0: irq 19, mapped at 9f80a000
[    3.040919] ftgmac100 1e680000.ethernet: Read MAC address 4a:3b:a5:b0:a2:a5 from chip
[    3.055841] libphy: ftgmac100_mdio: probed
[    3.124800] Broadcom BCM54612E 1e680000.ethernet--1:01: attached PHY driver [ Broadcom BCM54612E] (mii_bus:phy_addr=1e680000.ethernet--1:01, irq=-1)
[    3.139229] ftgmac100 1e680000.ethernet eth1: irq 20, mapped at 9f80c000
[    3.146214] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    3.152800] ehci-platform: EHCI generic platform driver
......
[FAILED] Failed to start Phosphor MBOX Daemon.
See 'systemctl status mboxd.service' for details.
[  OK  ] Started Phosphor Download Manager.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 romulus ttyS4

romulus login: ro[   63.209708] random: crng init done
ot
Password:
root@romulus:~#
root@romulus:~#
root@romulus:~#
root@romulus:~# [   93.123841] ------------[ cut here ]------------
[   93.128532] WARNING: CPU: 0 PID: 0 at /kernel-source//net/sched/sch_generic.c:316 dev_watchdog+0x22c/0x244
[   93.138190] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out
[   93.144978] CPU: 0 PID: 0 Comm: swapper Not tainted 4.13.16-101790466fa90d70d475ef015df8aa1527a23a73 #5
[   93.154362] Hardware name: Generic DT based system
[   93.159194] [<80109930>] (unwind_backtrace) from [<80106bc4>] (show_stack+0x20/0x24)
[   93.166958] [<80106bc4>] (show_stack) from [<8057eba4>] (dump_stack+0x20/0x28)
[   93.174201] [<8057eba4>] (dump_stack) from [<80115ed4>] (__warn+0xe0/0x108)
[   93.181180] [<80115ed4>] (__warn) from [<80115f50>] (warn_slowpath_fmt+0x54/0x74)
[   93.188681] [<80115f50>] (warn_slowpath_fmt) from [<8049e1e0>] (dev_watchdog+0x22c/0x244)
[   93.196879] [<8049e1e0>] (dev_watchdog) from [<80158088>] (call_timer_fn+0x40/0x124)
[   93.204634] [<80158088>] (call_timer_fn) from [<80158220>] (expire_timers+0xb4/0xc0)
[   93.212387] [<80158220>] (expire_timers) from [<80158320>] (run_timer_softirq+0xa4/0x19c)
[   93.220577] [<80158320>] (run_timer_softirq) from [<80101600>] (__do_softirq+0xd0/0x2e4)
[   93.228760] [<80101600>] (__do_softirq) from [<80119f9c>] (irq_exit+0xf4/0x13c)
[   93.236109] [<80119f9c>] (irq_exit) from [<8014c8c0>] (__handle_domain_irq+0x60/0xb8)
[   93.243956] [<8014c8c0>] (__handle_domain_irq) from [<80101524>] (avic_handle_irq+0x68/0x70)
[   93.252410] [<80101524>] (avic_handle_irq) from [<8010796c>] (__irq_svc+0x6c/0x90)
[   93.259978] Exception stack(0x80901ee8 to 0x80901f30)
[   93.265038] 1ee0:                   029468f7 00000000 ae627000 00000000 ffffe000 00000000
[   93.273210] 1f00: 80903070 8093f67e 00000001 806e3b48 80834a30 80901f44 80901f48 80901f38
[   93.281378] 1f20: 80102fc8 80102fc0 60000013 ffffffff
[   93.286449] [<8010796c>] (__irq_svc) from [<80102fc0>] (arch_cpu_idle+0x2c/0x3c)
[   93.293863] [<80102fc0>] (arch_cpu_idle) from [<80599f70>] (default_idle_call+0x30/0x3c)
[   93.301982] [<80599f70>] (default_idle_call) from [<80143eb8>] (do_idle+0x94/0xe4)
[   93.309572] [<80143eb8>] (do_idle) from [<801441a8>] (cpu_startup_entry+0x1c/0x20)
[   93.317159] [<801441a8>] (cpu_startup_entry) from [<80593f30>] (rest_init+0x90/0xb0)
[   93.324949] [<80593f30>] (rest_init) from [<80800e8c>] (start_kernel+0x358/0x3e4)
[   93.332436] ---[ end trace 192a7465d171bca7 ]---
root@romulus:~#
```
Build env and configuration:
```
Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-14.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "romulus"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm armv6 thumb arm1176jzs"
TARGET_FPU           = "soft"
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-perl            
meta-python          
meta-virtualization  
meta-phosphor        
meta-aspeed          
meta-ast2500         
meta-openpower       
meta-ibm             
meta-romulus         
workspace            = "master:10fcbdd532aca0d2a753abfc9a9914f2417927fc"
```
I modify the code and  enable MAC2 function.

ouyangxx10, 2018-04-26:
modify uboot configuration file and fix it.

closed 2018-04-26 by ouyangxx10

zpdsg123, 2022-08-10:
How do you modify the configuration?