openbmc/openbmc #3125: Clang code violations not failing CI

geissonator, 2018-04-24:
Not sure what happened but code that violates the repo clang format is not failing the CI job:

https://openpower.xyz/job/openbmc-repository-ci/13423/console

```
12:52:24 Formatting code under /home/jenkins-op/workspace/openbmc-repository-ci@2/openbmc/phosphor-state-manager/
12:52:25 diff --git a/chassis_state_manager.cpp b/chassis_state_manager.cpp
12:52:25 index 0180fd4..c0e338d 100644
12:52:25 --- a/chassis_state_manager.cpp
12:52:25 +++ b/chassis_state_manager.cpp
12:52:25 @@ -276,9 +276,8 @@ void Chassis::startPOHCounter()
12:52:25      phosphor::state::manager::EventPtr eventP{event};
12:52:25      event = nullptr;
12:52:25  
12:52:25 -    auto callback = [&]()
12:52:25 -    {
12:52:25 -        if (ChassisInherit::currentPowerState() ==  PowerState::On)
12:52:25 +    auto callback = [&]() {
12:52:25 +        if (ChassisInherit::currentPowerState() == PowerState::On)
12:52:25          {
12:52:25              pOHCounter(pOHCounter() + 1);
12:52:25          }
12:52:25 diff --git a/chassis_state_manager.hpp b/chassis_state_manager.hpp
12:52:25 index 1477b6b..2eec052 100644
12:52:25 --- a/chassis_state_manager.hpp
12:52:25 +++ b/chassis_state_manager.hpp
12:52:25 @@ -19,7 +19,7 @@ namespace POH
12:52:25  {
12:52:25  
12:52:25  using namespace std::chrono_literals;
12:52:25 -constexpr auto secondsPerHour = 3600s; //seconds Per Hour
12:52:25 +constexpr auto secondsPerHour = 3600s; // seconds Per Hour
12:52:25  
12:52:25  } // namespace  POH
12:52:25  
12:52:25 diff --git a/timer.cpp b/timer.cpp
12:52:25 index 1a7c879..4de6b3c 100644
12:52:25 --- a/timer.cpp
12:52:25 +++ b/timer.cpp
12:52:25 @@ -46,7 +46,7 @@ int Timer::timeoutHandler(sd_event_source* eventSource, uint64_t usec,
12:52:25      if (timer->getAction() == timer::ON)
12:52:25      {
12:52:25          auto r = sd_event_source_set_time(
12:52:25 -                     eventSource, (getTime() + timer->getDuration()).count());
12:52:25 +            eventSource, (getTime() + timer->getDuration()).count());
12:52:25          if (r < 0)
12:52:25          {
12:52:25              throw std::system_error(r, std::generic_category(), strerror(-r));
12:52:25 diff --git a/timer.hpp b/timer.hpp
12:52:25 index 1eabf62..5d5c7e3 100644
12:52:25 --- a/timer.hpp
12:52:25 +++ b/timer.hpp
12:52:25 @@ -107,13 +107,13 @@ class Timer
12:52:25      timer::Action action = timer::OFF;
12:52:25  
12:52:25      /** @brief Timer expiry handler - invokes callback
12:52:25 -    *
12:52:25 -    *  @param[in] eventSource - Source of the event
12:52:25 -    *  @param[in] usec        - time in micro seconds
12:52:25 -    *  @param[in] userData    - User data pointer
12:52:25 -    *
12:52:25 -    *  @return zero on success, non-zero otherwise
12:52:25 -    */
12:52:25 +     *
12:52:25 +     *  @param[in] eventSource - Source of the event
12:52:25 +     *  @param[in] usec        - time in micro seconds
12:52:25 +     *  @param[in] userData    - User data pointer
12:52:25 +     *
12:52:25 +     *  @return zero on success, non-zero otherwise
12:52:25 +     */
12:52:25      static int timeoutHandler(sd_event_source* eventSource, uint64_t usec,
12:52:25                                void* userData);
12:52:25  };
12:52:25 phosphor-state-manager
12:52:25 /home/jenkins-op/workspace/openbmc-repository-ci@2/openbmc/phosphor-state-manager > ./bootstrap.sh
12:52:25 + AUTOCONF_FILES=Makefile.in aclocal.m4 ar-lib autom4te.cache compile         config.guess config.h.in config.sub configure depcomp install-sh         ltmain.sh missing *libtool test-driver
12:52:25 + autoreconf -i
12:52:29 libtoolize: putting auxiliary files in '.'.
12:52:29 libtoolize: copying file './ltmain.sh'
12:52:29 libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
12:52:29 libtoolize: and rerunning libtoolize and aclocal.
12:52:29 libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
12:52:33 configure.ac:27: installing './compile'
...
```

geissonator, 2018-04-24:
FYI @amboar  I haven't had time to dig in yet.  It would appear the "git --no-pager diff --exit-code" command is not returning a non-zero code anymore?

amboar, 2018-04-29:
Hmm. I haven't had time to dig in either. I'll try to get to this today

amboar, 2018-04-30:
Ok, this looks straight-forward - shell scripts exit with the exit status of the last command:

```
2 11:31:33 andrew@dave:/tmp$ cat <<EOF > good.sh
> false
> true
> EOF
2 11:33:49 andrew@dave:/tmp$ bash ./good.sh 
2 11:33:55 andrew@dave:/tmp$ echo $?
0
2 11:33:56 andrew@dave:/tmp$ cat <<EOF > evil.sh
> true
> false
> EOF
2 11:34:05 andrew@dave:/tmp$ bash evil.sh
2 11:34:11 andrew@dave:/tmp$ echo $?
1
2 11:34:13 andrew@dave:/tmp$ 
```
Adding code to the bottom of the script made it unexpectedly pass.  I'll send a patch adding `set -e`, though we'll need to clean up the mess again :/

amboar, 2018-04-30:
I've got a few fixes so far:

https://gerrit.openbmc-project.xyz/c/10374/
https://gerrit.openbmc-project.xyz/c/10373/

rfrandse, 2018-04-30:
<https://gerrit.openbmc-project.xyz/10378> scripts/format-code.sh: Immediately exit on bad exit status
Fixes  #3125 Clang code violations not failing CI

closed 2018-05-01 by geissonator