openbmc/openbmc #3129: witherspoon: phosphor-fan-control 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'

gkeishin, 2018-04-26:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.2-35"
VERSION_ID="v2.2-35-geee4542"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.2-35"
BUILD_ID="v2.2"
root@witherspoon:~#
```

Journald snippet
```
Apr 26 06:59:27 witherspoon systemd[1]: Started Witherspoon UART renderer.
Apr 26 06:59:28 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Main process exited, code=dumped, status=6/ABRT
Apr 26 06:59:29 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Unit entered failed state.
Apr 26 06:59:29 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Failed with result 'core-dump'.
Apr 26 06:59:30 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Service hold-off time over, scheduling restart.
Apr 26 06:59:30 witherspoon systemd[1]: Stopped Phosphor Fan Control Initialization.
Apr 26 06:59:30 witherspoon systemd[1]: Started Phosphor Fan Control Initialization.
Apr 26 06:59:31 witherspoon phosphor-fan-control[1718]: Failed to invoke DBus method.
Apr 26 06:59:31 witherspoon phosphor-fan-control[1718]: The operation failed internally.
Apr 26 06:59:31 witherspoon phosphor-fan-control[1718]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Apr 26 06:59:31 witherspoon phosphor-fan-control[1718]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Apr 26 06:59:31 witherspoon systemd[1]: Started Process Core Dump (PID 1719/UID 0).
Apr 26 06:59:31 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Main process exited, code=dumped, status=6/ABRT
Apr 26 06:59:31 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Unit entered failed state.
Apr 26 06:59:31 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Failed with result 'core-dump'.
Apr 26 06:59:32 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Service hold-off time over, scheduling restart.
```

- [x] Journald log  
[journald.txt](https://github.com/openbmc/openbmc/files/1950076/journald.txt)

- [x] dmesg log 
[dmesg.txt](https://github.com/openbmc/openbmc/files/1950067/dmesg.txt)

gkeishin, 2018-04-26:
```
[  607.717112] systemd[1]: Started Process Core Dump (PID 2054/UID 0).
[  642.047457] systemd[1]: Starting Hostname Service...
[  643.781947] systemd[1]: Started Hostname Service.
[  647.625104] systemd[1]: systemd-journald.service: State 'stop-sigabrt' timed out. Terminating.
[  657.307757] systemd[1]: systemd-coredump@5-1734-0.service: Service reached runtime time limit. Stopping.
[  665.699116] systemd[1]: systemd-coredump@6-1737-0.service: Service reached runtime time limit. Stopping.
[  666.968280] systemd[1]: systemd-coredump@5-1734-0.service: Unit entered failed state.
[  667.009124] systemd[1]: systemd-coredump@5-1734-0.service: Failed with result 'timeout'.
[  667.130216] systemd[1]: systemd-coredump@6-1737-0.service: Unit entered failed state.
[  667.138513] systemd[1]: systemd-coredump@6-1737-0.service: Failed with result 'timeout'.
[  677.068658] systemd[1]: Starting Hostname Service...
[  677.747546] systemd[1]: systemd-journald.service: Main process exited, code=dumped, status=6/ABRT
[  677.933967] systemd[1]: systemd-journald.service: Unit entered failed state.
[  677.943569] systemd[1]: systemd-journald.service: Failed with result 'watchdog'.
[  678.041675] systemd[1]: systemd-journald.service: Service has no hold-off time, scheduling restart.
[  678.849118] systemd[1]: Stopped Flush Journal to Persistent Storage.
[  678.856055] systemd[1]: Stopping Flush Journal to Persistent Storage...
[  678.863192] systemd[1]: Stopped Journal Service.
[  679.028799] systemd[1]: Starting Journal Service...
[  679.166373] systemd[1]: phosphor-fan-control-init@0.service: Main process exited, code=dumped, status=6/ABRT
[  684.513777] systemd[1]: Started Process Core Dump (PID 2476/UID 0).
[  685.113992] systemd-journald[2416]: File /var/log/journal/1b223cb79c3e4d6091db820a12154cf3/system.journal corrupted or uncleanly shut down, renaming and replacing.
[  685.804437] systemd[1]: Started Journal Service.
[  789.913461] systemd-coredump[1900]: MESSAGE=Process 685 (systemd-journal) of user 0 dumped core.
[  789.922352] systemd-coredump[1900]: Coredump diverted to /var/lib/systemd/coredump/core.systemd-journal.0.6560214cd335489cbae063a85ab75c6b.685.1524726197000000.xz
[  976.949592] UBIFS warning (ubi0:2 pid 2416): ubifs_vm_page_mkwrite: out of space for mmapped file (inode number 3562)
[  976.960599] UBIFS warning (ubi0:2 pid 2416): ubifs_vm_page_mkwrite: out of space for mmapped file (inode number 3562)
[  977.070107] systemd-journald[2416]: /var/log/journal/1b223cb79c3e4d6091db820a12154cf3/system.journal: Journal file corrupted, rotating.
[ 1155.138316] UBIFS warning (ubi0:2 pid 2416): ubifs_vm_page_mkwrite: out of space for mmapped file (inode number 3689)
[ 1155.149376] UBIFS warning (ubi0:2 pid 2416): ubifs_vm_page_mkwrite: out of space for mmapped file (inode number 3689)
[ 1155.160367] systemd-journald[2416]: /var/log/journal/1b223cb79c3e4d6091db820a12154cf3/system.journal: Journal file corrupted, rotating.
[ 1155.172938] systemd-journald[2416]: Failed to rotate /var/log/journal/1b223cb79c3e4d6091db820a12154cf3/system.journal: No space left on device
[ 1155.189199] systemd-journald[2416]: Failed to write entry (24 items, 783 bytes) despite vacuuming, ignoring: Input/output error
```
```
root@witherspoon:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                17.3M     17.3M         0 100% /
devtmpfs                193.7M         0    193.7M   0% /dev
ubi0:rwfs                 4.5M      3.9M    416.0K  91% /var
overlay                   4.5M      3.9M    416.0K  91% /etc
tmpfs                   210.2M         0    210.2M   0% /dev/shm
tmpfs                   210.2M    760.0K    209.5M   0% /run
tmpfs                   210.2M         0    210.2M   0% /sys/fs/cgroup
tmpfs                   210.2M      2.4M    207.8M   1% /tmp
tmpfs                   210.2M         0    210.2M   0% /media
tmpfs                   210.2M         0    210.2M   0% /var/volatile
ubi0:rwfs                 4.5M      3.9M    416.0K  91% /home
tmpfs                   210.2M         0    210.2M   0% /usr/local
tmpfs                   210.2M         0    210.2M   0% /srv
/dev/ubiblock8_0         22.3M     22.3M         0 100% /media/pnor-ro-031bc06e
/dev/ubiblock0_0         17.3M     17.3M         0 100% /media/rofs-5b607635
/dev/ubiblock8_1         22.4M     22.4M         0 100% /media/pnor-ro-7d06e417
ubi8:pnor-prsv          972.0K     32.0K    868.0K   4% /media/pnor-prsv
ubi8:pnor-rw-031bc06e
                         13.5M    260.0K     12.5M   2% /media/pnor-rw-031bc06e
ubi8:pnor-rw-7d06e417
                         13.5M    316.0K     12.5M   2% /media/pnor-rw-7d06e417
ubi8:pnor-patch          13.5M     16.0K     12.7M   0% /usr/local/share/pnor
/dev/ubiblock4_0         16.6M     16.6M         0 100% /media/rofs-a2228808
root@witherspoon:~#
```

gkeishin, 2018-04-26:
On discussion with @spinler 
the /etc overlay had some left over stuff from a CI run that had my changes: lrwxrwxrwx    1 root     root            77 Apr 25 22:56 max31785@52.conf -> /etc/default/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@100/max31785@52_air.conf

msbarth, 2018-04-26:
Seems this issue can be closed now?

gkeishin, 2018-04-27:
sure thing.. will close it up then..

closed 2018-04-27 by gkeishin