openbmc/openbmc #3131: Witherspoon - Services application core dump continuously filling up space

gkeishin, 2018-04-26:
If a service application continuously  core dumps,  we have the rule to check for a certain amount of time 10-second x core dumping and stop the service to avoid filling up space on BMC

However we hit a scenario as per @geissonator , the problem here is the service has to fail twice within 10 seconds, then systemd will stop starting it.  This one is failing outside that 10s boundary so it keeps getting restarted:

```
Apr 26 07:18:12 witherspoon systemd[1]: Started Process Core Dump (PID 4442/UID 0).
Apr 26 07:18:27 witherspoon systemd[1]: Started Process Core Dump (PID 4517/UID 0).
Apr 26 07:18:35 witherspoon systemd[1]: Started Process Core Dump (PID 4520/UID 0).
Apr 26 07:18:57 witherspoon systemd[1]: Started Process Core Dump (PID 4613/UID 0).
Apr 26 07:19:25 witherspoon systemd[1]: Started Process Core Dump (PID 4903/UID 0).
Apr 26 11:45:59 witherspoon systemd[1]: Started Process Core Dump (PID 5677/UID 0).
Apr 26 12:23:20 witherspoon systemd[1]: Started Process Core Dump (PID 6047/UID 0).
Apr 26 12:23:32 witherspoon systemd[1]: Started Process Core Dump (PID 6211/UID 0).
```

geissonator, 2018-04-26:
My vote is we bump up to a 30s limit. In general, if the first restart doesn't work, I don't see us getting much from further restarts.

rfrandse, 2018-04-26:
<https://gerrit.openbmc-project.xyz/10332> Change the StartLimitIntervalSec to 30s
Resolves  #3131 Witherspoon - Services application core dump continuously filling up space

closed 2018-05-02 by geissonator