openbmc/openbmc #3135: phosphor-net-ipmid fails to compile with OpenSSL 1.1.0

amboar, 2018-04-30:
OpenSSL 1.1.0 (is shipped in Ubuntu 18.04) contains a change of interface for the EVP_EncryptInit suite. Comparing the man pages:

Ubuntu Artful (17.10): http://manpages.ubuntu.com/manpages/artful/man3/EVP_EncryptInit.3ssl.html
Ubuntu Bionic (18.04): http://manpages.ubuntu.com/manpages/bionic/man3/EVP_EncryptInit.3ssl.html

Particularly, `diff`ing the examples (with some cleanup) gives:

```
--- /tmp/artful.c	2018-04-30 13:22:23.327354618 +0930
+++ /tmp/bionic.c	2018-04-30 13:21:55.483568260 +0930
@@ -8,30 +8,30 @@
        unsigned char key[] = {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15};
        unsigned char iv[] = {1,2,3,4,5,6,7,8};
        char intext[] = "Some Crypto Text";
-       EVP_CIPHER_CTX ctx;
+       EVP_CIPHER_CTX *ctx;
        FILE *out;
 
-       EVP_CIPHER_CTX_init(&ctx);
-       EVP_EncryptInit_ex(&ctx, EVP_idea_cbc(), NULL, key, iv);
+       ctx = EVP_CIPHER_CTX_new();
+       EVP_EncryptInit_ex(ctx, EVP_idea_cbc(), NULL, key, iv);
 
-       if(!EVP_EncryptUpdate(&ctx, outbuf, &outlen, intext, strlen(intext)))
+       if(!EVP_EncryptUpdate(ctx, outbuf, &outlen, intext, strlen(intext)))
        {
              /* Error */
              return 0;
        }
        /* Buffer passed to EVP_EncryptFinal() must be after data just
-       * encrypted to avoid overwriting it.
-       */
-       if(!EVP_EncryptFinal_ex(&ctx, outbuf + outlen, &tmplen))
+        * encrypted to avoid overwriting it.
+        */
+       if(!EVP_EncryptFinal_ex(ctx, outbuf + outlen, &tmplen))
        {
              /* Error */
              return 0;
        }
        outlen += tmplen;
-       EVP_CIPHER_CTX_cleanup(&ctx);
+       EVP_CIPHER_CTX_free(ctx);
        /* Need binary mode for fopen because encrypted data is
         * binary data. Also cannot use strlen() on it because
-        * it wont be null terminated and may contain embedded
+        * it won't be null terminated and may contain embedded
         * nulls.
         */
        out = fopen(outfile, "wb");
```
We will need to abstract over this breaking interface change.

anoo1, 2018-09-06:
The error is:
```
crypt_algo.cpp: In member function 'std::vector<unsigned char> cipher::crypt::AlgoAES128::decryptData(const uint8_t*, const uint8_t*, int) const':^M
crypt_algo.cpp:93:20: error: aggregate 'EVP_CIPHER_CTX ctx' has incomplete type and cannot be defined^M
     EVP_CIPHER_CTX ctx;
                    ^~~
crypt_algo.cpp: In member function 'std::vector<unsigned char> cipher::crypt::AlgoAES128::encryptData(const uint8_t*, int) const':^M
crypt_algo.cpp:159:20: error: aggregate 'EVP_CIPHER_CTX ctx' has incomplete type and cannot be defined^M
     EVP_CIPHER_CTX ctx;
```

rfrandse, 2018-09-07:
<https://gerrit.openbmc-project.xyz/12559> Add support for OpenSSL 1.1.0
Resolves  #3135 phosphor-net-ipmid fails to compile with OpenSSL 1.1.0

closed 2018-09-10 by geissonator