openbmc/openbmc #3136: phosphor-bmc-code-mgmt fails to compile with OpenSSL 1.1.0

amboar, 2018-04-30:
phosphor-bmc-code-mgmt fails to compile with OpenSSL 1.1.0, which is shipped in Ubuntu Bionic (18.04):

```
In file included from item_updater.cpp:15:0:    
image_verify.hpp:29:45: error: '::EVP_MD_CTX_destroy' has not been declared                      
     std::unique_ptr<EVP_MD_CTX, decltype(&::EVP_MD_CTX_destroy)>;                               
                                             ^~~~~~~~~~~~~~~~~~
image_verify.hpp:29:45: note: suggested alternative: 'EVP_MD_CTX_reset'                          
     std::unique_ptr<EVP_MD_CTX, decltype(&::EVP_MD_CTX_destroy)>;                               
                                             ^~~~~~~~~~~~~~~~~~
                                             EVP_MD_CTX_reset
image_verify.hpp:29:64: error: template argument 2 is invalid                                    
     std::unique_ptr<EVP_MD_CTX, decltype(&::EVP_MD_CTX_destroy)>;                               
                                                                ^
```

gtmills, 2018-09-04:
We moved our docker image to Ubuntu Bionic and seeing this fail in phosphor-bmc-code-mgmt CI.

gtmills, 2018-09-04:
```
'/usr/local/sbin'
15:07:18 libtool: install: /usr/bin/install -c phosphor-version-software-manager /usr/local/sbin/phosphor-version-software-manager
15:07:18 libtool: install: /usr/bin/install -c phosphor-download-manager /usr/local/sbin/phosphor-download-manager
15:07:18 libtool: install: /usr/bin/install -c phosphor-image-updater /usr/local/sbin/phosphor-image-updater
15:07:18 make[3]: Leaving directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/phosphor-bmc-code-mgmt'
15:07:18 /var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/phosphor-bmc-code-mgmt > ldconfig
15:07:18 /var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/phosphor-bmc-code-mgmt > make -j 16 -l 16 -O check
15:07:18 make  check-recursive
15:07:18 Making check in test
15:07:18 make  utest
15:07:20 make[3]: Entering directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/phosphor-bmc-code-mgmt/test'
15:07:20 g++ -DHAVE_CONFIG_H -I. -I..  -Igtest -DGTEST_HAS_PTHREAD=1 -I..  -pthread -I/usr/local/include -I/usr/local/include -DUBIFS_LAYOUT -g -O2 -Wall -Werror -MT utest-utest.o -MD -MP -MF .deps/utest-utest.Tpo -c -o utest-utest.o `test -f 'utest.cpp' || echo './'`utest.cpp
15:07:20 In file included from utest.cpp:10:0:
15:07:20 ../image_verify.hpp:29:45: error: '::EVP_MD_CTX_destroy' has not been declared
15:07:20      std::unique_ptr<EVP_MD_CTX, decltype(&::EVP_MD_CTX_destroy)>;
15:07:20                                              ^~~~~~~~~~~~~~~~~~
15:07:20 ../image_verify.hpp:29:45: note: suggested alternative: 'EVP_MD_CTX_reset'
15:07:20      std::unique_ptr<EVP_MD_CTX, decltype(&::EVP_MD_CTX_destroy)>;
15:07:20                                              ^~~~~~~~~~~~~~~~~~
15:07:20                                              EVP_MD_CTX_reset
15:07:20 ../image_verify.hpp:29:64: error: template argument 2 is invalid
15:07:20      std::unique_ptr<EVP_MD_CTX, decltype(&::EVP_MD_CTX_destroy)>;
15:07:20                                                                 ^
15:07:20 Makefile:633: recipe for target 'utest-utest.o' failed
15:07:20 make[3]: *** [utest-utest.o] Error 1
15:07:20 make[3]: Leaving directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/phosphor-bmc-code-mgmt/test'
15:07:20 Makefile:898: recipe for target 'check-am' failed
15:07:20 make[2]: *** [check-am] Error 2
15:07:20 Makefile:1041: recipe for target 'check-recursive' failed
15:07:20 make[1]: *** [check-recursive] Error 1
15:07:20 Makefile:1329: recipe for target 'check' failed
15:07:20 make: *** [check] Error 2
```

rfrandse, 2018-09-06:
<https://gerrit.openbmc-project.xyz/12536> image_verify: Add support for OpenSSL 1.1.0
Resolves  #3136 phosphor-bmc-code-mgmt fails to compile with OpenSSL 1.1.0

closed 2018-09-12 by geissonator