openbmc/openbmc #3140: settings_manager.py from phosphor-settingsd contains invalid regex escape

amboar, 2018-04-30:
openbmc/phosphor-settingsd@a16abc5aa2d30b9605e30ffd6bbe3aa05585ae87 contains an invalid regex escape in settings_manager.py:

```
+ docker run --cap-add=sys_admin --rm=true --privileged=true -w /tmp/openbmc-build-scripts.DlARM6 -v /tmp/openbmc-build-scripts.DlARM6:/tmp/openbmc-build-scripts.DlARM6 -e 'MAKEFLAGS=-j 4' -t openbmc/ubuntu-unit-test /tmp/openbmc-build-scripts.DlARM6/dbus-unit-test.py -u /tmp/openbmc-build-scripts.DlARM6/unit-test.py,-w,/tmp/openbmc-build-scripts.DlARM6,-p,phosphor-settingsd,-v -f /usr/share/dbus-1/system.conf
/tmp/openbmc-build-scripts.DlARM6 > ./format-code.sh /tmp/openbmc-build-scripts.DlARM6/phosphor-settingsd
Formatting code under /tmp/openbmc-build-scripts.DlARM6/phosphor-settingsd/
./settings_manager.py:221:36: W605 invalid escape sequence '\-'
                regex = '([a-fA-F0-9]{2}[:|\-]?){6}'
                                   ^
Traceback (most recent call last):
  File "/tmp/openbmc-build-scripts.DlARM6/unit-test.py", line 409, in <module>
    check_call_cmd(WORKSPACE, "./format-code.sh", CODE_SCAN_DIR)
  File "/tmp/openbmc-build-scripts.DlARM6/unit-test.py", line 224, in check_call_cmd
    check_call(cmd)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '('./format-code.sh', '/tmp/openbmc-build-scripts.DlARM6/phosphor-settingsd')' returned non-zero exit status 1
Traceback (most recent call last):
  File "/tmp/openbmc-build-scripts.DlARM6/dbus-unit-test.py", line 91, in <module>
    check_call(UNIT_TEST.split(','), env=os.environ)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/tmp/openbmc-build-scripts.DlARM6/unit-test.py', '-w', '/tmp/openbmc-build-scripts.DlARM6', '-p', 'phosphor-settingsd', '-v']' returned non-zero exit status 1
```

amboar, 2018-04-30:
I've pushed a fix: https://gerrit.openbmc-project.xyz/#/c/10373/

rfrandse, 2018-04-30:
<https://gerrit.openbmc-project.xyz/10373> settings_manager: Use raw string for regex
Fixes  #3140 settings_manager.py from phosphor-settingsd contains invalid escape sequence

closed 2018-04-30 by geissonator