openbmc/openbmc #3154: Rest server should return a proper JSON bool rather than a string of "1" or "0"

gtmills, 2018-05-03:
```$ curl -c cjar -b cjar -k  https://${BMC_IP}/xyz/openbmc_project/network/eth0/attr/DHCPEnabled
{
  "data": 0, 
  "message": "200 OK", 
  "status": "ok"
}
```

gtmills, 2018-05-23:
https://github.com/openbmc/openbmc/issues/554  is a related story

spinler, 2018-05-24:
To level set, this is how dbus.Boolean works.

From https://dbus.freedesktop.org/doc/dbus-python/dbus.html#dbus.Boolean:

```
 class dbus.Boolean(value: bool[, variant_level: int])

    Bases: _dbus_bindings._IntBase
    A boolean, represented as a subtype of int (not bool, because bool cannot be subclassed).
    value is converted to 0 or 1 as if by int(bool(value)).
```

spinler, 2018-05-24:
The json module that Bottle uses does let you extend the encoder for types it doesn't understand, but it understands dbus.Boolean perfectly fine - it thinks it's an int:

```
>>> value = dbus.Boolean(True)
>>> print value
1
>>> isinstance(value, int)
True
```
So it won't even call the custom encoder.

rfrandse, 2018-06-25:
<https://gerrit.openbmc-project.xyz/11267> Explicitly convert dbus.Booleans to bools
Resolves  #3154 Rest server should return a proper JSON bool rather than a string of "1" or "0"

spinler, 2018-06-28:
I had to resort to introspecting all JSON responses in rest_dbus.py and explicitly doing the conversions.

whowutwut, 2018-07-16:
@gtmills Is that output in the description the new behavior? Or the old one? 
It doesn't look like it's returning as as string today... 

How are the new changes different? 

Running the following FW:
```
[root@briggs01 Network]# rinv mid05tor12cn15 firm
mid05tor12cn15: BMC Firmware Product: ibm-v2.1-438-g0030304-r15-0-g19832d3 (Active)*
mid05tor12cn15: Host Firmware Product: IBM-witherspoon-ibm-OP9-v2.0.3-2.17 (Active)*
mid05tor12cn15: Host Firmware Product: -- additional info: buildroot-2018.02.1-6-ga8d1126
mid05tor12cn15: Host Firmware Product: -- additional info: capp-ucode-p9-dd2-v4
mid05tor12cn15: Host Firmware Product: -- additional info: hcode-fbe2c7c
mid05tor12cn15: Host Firmware Product: -- additional info: hostboot-9df0950-p378630b
mid05tor12cn15: Host Firmware Product: -- additional info: hostboot-binaries-8097b96
mid05tor12cn15: Host Firmware Product: -- additional info: linux-4.16.13-openpower1-pc10362e
mid05tor12cn15: Host Firmware Product: -- additional info: machine-xml-7cd20a6
mid05tor12cn15: Host Firmware Product: -- additional info: occ-90208bb
mid05tor12cn15: Host Firmware Product: -- additional info: op-build-v2.0.3-2-g6279a04-dirty
mid05tor12cn15: Host Firmware Product: -- additional info: petitboot-v1.7.1-pcc0e2c7
mid05tor12cn15: Host Firmware Product: -- additional info: sbe-0b006e7
mid05tor12cn15: Host Firmware Product: -- additional info: skiboot-v6.0.4
```

And printing the DHCPEnabled attribute returns 0.. 
```
[root@briggs01 Network]# curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://172.11.139.15/xyz/openbmc_project/network/eth0_11/attr/DHCPEnabled
{
  "data": 0,
  "message": "200 OK",
  "status": "ok"
```

spinler, 2018-07-16:
@whowutwut 0 is the old behavior, the new behavior will be: false

closed 2018-08-28 by geissonator