openbmc/openbmc #3164: OP920: Kernel CRTC: vblank wait timed out

cdeadmin, 2018-05-07:


gkeishin, 2018-05-07:
```
dmesg with debug enabled

root@witherspoon:~# dmesg
[   46.285314] ubi8: attached mtd8 (name "pnor", size 128 MiB)
[   46.290916] ubi8: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[   46.297533] ubi8: min./max. I/O unit sizes: 1/256, sub-page size 1
[   46.303711] ubi8: VID header offset: 64 (aligned 64), data offset: 128
[   46.310237] ubi8: good PEBs: 2048, bad PEBs: 0, corrupted PEBs: 0
[   46.316330] ubi8: user volume: 6, internal volumes: 1, max. volumes count: 128
[   46.323546] ubi8: max/mean erase counter: 76/56, WL threshold: 4096, image sequence number: 1771973205
[   46.332840] ubi8: available PEBs: 520, total reserved PEBs: 1528, PEBs reserved for bad PEB handling: 0
[   46.342291] ubi8: background thread "ubi_bgt8d" started, PID 1133
[   50.380829] 8021q: adding VLAN 0 to HW filter on device eth0
[   50.410729] ftgmac100 1e660000.ethernet eth0: NCSI: Handler for packet type 0x82 returned -19
[   53.712793] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
[   53.782165] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
[   53.782202] ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
[   53.782221] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[   58.863925] block ubiblock8_0: created from ubi8:0(pnor-ro-9a2dfeba)
[   59.873844] UBIFS (ubi8:1): UBIFS: mounted UBI device 8, volume 1, name "pnor-rw-9a2dfeba"
[   59.882334] UBIFS (ubi8:1): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   59.891749] UBIFS (ubi8:1): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[   59.902095] UBIFS (ubi8:1): reserved for root: 763076 bytes (745 KiB)
[   59.908540] UBIFS (ubi8:1): media format: w5/r0 (latest is w5/r0), UUID 6AEDAF8D-929F-4CC3-A9F6-BC3DF18DA1B4, small LPT model
[   59.923287] UBIFS (ubi8:1): background thread "ubifs_bgt8_1" started, PID 1162
[   60.613435] UBIFS (ubi8:2): background thread "ubifs_bgt8_2" started, PID 1178
[   60.698905] UBIFS (ubi8:2): UBIFS: mounted UBI device 8, volume 2, name "pnor-prsv"
[   60.706592] UBIFS (ubi8:2): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   60.715985] UBIFS (ubi8:2): FS size: 1504384 bytes (1 MiB, 23 LEBs), journal size 523265 bytes (0 MiB, 6 LEBs)
[   60.726156] UBIFS (ubi8:2): reserved for root: 71055 bytes (69 KiB)
[   60.732443] UBIFS (ubi8:2): media format: w5/r0 (latest is w5/r0), UUID EC01E1C8-9696-4758-8BE0-6B74670C9409, small LPT model
[   73.933382] block ubiblock8_4: created from ubi8:4(pnor-ro-bfc0037c)
[   74.321911] UBIFS (ubi8:5): UBIFS: mounted UBI device 8, volume 5, name "pnor-rw-bfc0037c"
[   74.330402] UBIFS (ubi8:5): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   74.339957] UBIFS (ubi8:5): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[   74.350318] UBIFS (ubi8:5): reserved for root: 763076 bytes (745 KiB)
[   74.356769] UBIFS (ubi8:5): media format: w5/r0 (latest is w5/r0), UUID 385278F4-DD31-49DB-95BB-E20D3A780F81, small LPT model
[   74.373641] UBIFS (ubi8:5): background thread "ubifs_bgt8_5" started, PID 1217
[   75.604393] block ubiblock0_1: created from ubi0:1(rofs-b930f218)
[   97.119109] ftgmac100 1e660000.ethernet eth0: Set package 0x0, channel 0x0 as preferred
[   97.127484] ftgmac100 1e660000.ethernet eth0: NCSI: Stopping device
[   97.127508] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[   97.133671] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
[   97.144419] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
[   97.144454] ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
[   97.144510] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[  107.253127] UBIFS (ubi8:3): background thread "ubifs_bgt8_3" started, PID 1337
[  107.276444] UBIFS (ubi8:3): UBIFS: mounted UBI device 8, volume 3, name "pnor-patch"
[  107.284222] UBIFS (ubi8:3): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[  107.293619] UBIFS (ubi8:3): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[  107.303961] UBIFS (ubi8:3): reserved for root: 763076 bytes (745 KiB)
[  107.310406] UBIFS (ubi8:3): media format: w5/r0 (latest is w5/r0), UUID C48E4DC3-D116-4CB0-B4B1-886F0D41A94D, small LPT model
[  154.573620] ftgmac100 1e660000.ethernet eth0: Set package 0x0, channel 0x0 as preferred
[  154.581663] ftgmac100 1e660000.ethernet eth0: NCSI: Stopping device
[  154.581681] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[  154.587787] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
[  154.598225] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
[  154.598259] ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
[  154.598276] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[  190.068567] occ-hwmon occ-hwmon.1: failed to get OCC poll response: -19
[  190.606203] occ-hwmon occ-hwmon.2: failed to get OCC poll response: -19
[ 1012.011628] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 1017.636692] occ-hwmon occ-hwmon.1: failed to get OCC poll response: -19
[ 1018.481194] occ-hwmon occ-hwmon.2: failed to get OCC poll response: -19
[ 1024.495919] [drm:drm_atomic_state_init] Allocated atomic state 93aab940
[ 1024.495960] [drm:drm_atomic_get_plane_state] Added [PLANE:28:plane-0] 97da2c80 state to 93aab940
[... cut ...]
[ 1120.068467] [drm:drm_atomic_state_default_clear] Clearing atomic state 91128ec0
[ 1120.068525] [drm:__drm_atomic_state_free] Freeing atomic state 91128ec0
[ 1122.015294] ftgmac100 1e660000.ethernet eth0: NCSI: LSC AEN - channel 0 state up
[ 1122.022956] ftgmac100 1e660000.ethernet eth0: NCSI: host driver running on channel 0
[ 1134.618566] ftgmac100 1e660000.ethernet eth0: NCSI: host driver not running on channel 0
[ 1163.618031] ftgmac100 1e660000.ethernet eth0: NCSI: host driver running on channel 0
root@witherspoon:~#
```

gkeishin, 2018-05-08:
Another failure dmesg from different run
```
root@witherspoon:~# dmesg
[   21.054340] systemd[1]: Failed to insert module 'autofs4': No such file or directory
[   21.264758] systemd[1]: systemd 234 running in system mode. (+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN2 -IDN default-hierarchy=hybrid)
[   21.286949] systemd[1]: Detected architecture arm.
[   21.347952] systemd[1]: Set hostname to <witherspoon>.
[   23.233972] systemd[1]: Created slice System Slice.
[   23.269936] systemd[1]: Created slice system-phosphor\x2dreset\x2dsensor\x2dstates.slice.
[   23.309575] systemd[1]: Created slice system-org.openbmc.buttons.Power.slice.
[   23.350896] systemd[1]: Created slice system-phosphor\x2ddiscover\x2dsystem\x2dstate.slice.
[   23.409537] systemd[1]: Created slice system-phosphor\x2dreset\x2dhost\x2dcheck.slice.
[   23.447783] systemd[1]: Reached target Remote File Systems.
[   23.489566] systemd[1]: Created slice system-org.openbmc.control.Power.slice.
[   26.588798] systemd[1]: Started Load/Save Random Seed.
[   26.832066] systemd[1]: Started udev Kernel Device Manager.
[   26.858641] systemd[1]: Started Bind mount volatile /srv.
[   26.874563] systemd[1]: Reached target Local File Systems.
[   27.024441] systemd[1]: Starting Journal Service...
[   28.743659] systemd[1]: Started udev Coldplug all Devices.
[   28.866452] systemd[1]: Found device /dev/ttyS4.
[   29.318409] systemd[1]: Started Journal Service.
[   33.366825] systemd-journald[684]: Received request to flush runtime journal from PID 1
[   38.559181] ubi8: default fastmap pool size: 100
[   38.563829] ubi8: default fastmap WL pool size: 50
[   38.568621] ubi8: attaching mtd8
[   38.598260] ubi8: scanning is finished
[   38.742714] ubi8: attached mtd8 (name "pnor", size 128 MiB)
[   38.748317] ubi8: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[   38.754938] ubi8: min./max. I/O unit sizes: 1/256, sub-page size 1
[   38.761114] ubi8: VID header offset: 64 (aligned 64), data offset: 128
[   38.767632] ubi8: good PEBs: 2048, bad PEBs: 0, corrupted PEBs: 0
[   38.773717] ubi8: user volume: 4, internal volumes: 1, max. volumes count: 128
[   38.780939] ubi8: max/mean erase counter: 77/57, WL threshold: 4096, image sequence number: 1771973205
[   38.790235] ubi8: available PEBs: 1136, total reserved PEBs: 912, PEBs reserved for bad PEB handling: 0
[   38.799680] ubi8: background thread "ubi_bgt8d" started, PID 1031
[   45.957165] block ubiblock8_0: created from ubi8:0(pnor-ro-3d8a0973)
[   46.584024] UBIFS (ubi8:1): UBIFS: mounted UBI device 8, volume 1, name "pnor-rw-3d8a0973"
[   46.592322] UBIFS (ubi8:1): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   46.601715] UBIFS (ubi8:1): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[   46.612050] UBIFS (ubi8:1): reserved for root: 763076 bytes (745 KiB)
[   46.618496] UBIFS (ubi8:1): media format: w5/r0 (latest is w5/r0), UUID 29464610-6132-4B59-82C8-7029D9F3D2CD, small LPT model
[   46.633029] UBIFS (ubi8:1): background thread "ubifs_bgt8_1" started, PID 1050
[   47.101281] UBIFS (ubi8:2): UBIFS: mounted UBI device 8, volume 2, name "pnor-prsv"
[   47.108968] UBIFS (ubi8:2): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   47.118526] UBIFS (ubi8:2): FS size: 1504384 bytes (1 MiB, 23 LEBs), journal size 523265 bytes (0 MiB, 6 LEBs)
[   47.128535] UBIFS (ubi8:2): reserved for root: 71055 bytes (69 KiB)
[   47.134805] UBIFS (ubi8:2): media format: w5/r0 (latest is w5/r0), UUID EC01E1C8-9696-4758-8BE0-6B74670C9409, small LPT model
[   47.152275] UBIFS (ubi8:2): background thread "ubifs_bgt8_2" started, PID 1058
[   48.469330] 8021q: adding VLAN 0 to HW filter on device eth0
[   48.486471] ftgmac100 1e660000.ethernet eth0: NCSI: Handler for packet type 0x82 returned -19
[   52.044162] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
[   52.188511] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
[   52.188546] ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
[   52.188563] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[   70.417388] block ubiblock0_1: created from ubi0:1(rofs-00efdb5a)
[   89.169436] ftgmac100 1e660000.ethernet eth0: Set package 0x0, channel 0x0 as preferred
[   89.177502] ftgmac100 1e660000.ethernet eth0: NCSI: Stopping device
[   89.177518] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[   89.183647] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
[   89.205240] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
[   89.205273] ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
[   89.205291] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[   95.353396] UBIFS (ubi8:3): background thread "ubifs_bgt8_3" started, PID 1218
[   95.450013] UBIFS (ubi8:3): UBIFS: mounted UBI device 8, volume 3, name "pnor-patch"
[   95.457790] UBIFS (ubi8:3): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   95.467183] UBIFS (ubi8:3): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[   95.477712] UBIFS (ubi8:3): reserved for root: 763076 bytes (745 KiB)
[   95.484178] UBIFS (ubi8:3): media format: w5/r0 (latest is w5/r0), UUID C48E4DC3-D116-4CB0-B4B1-886F0D41A94D, small LPT model
[... cut ...]
[  303.434920] [drm:drm_atomic_state_default_clear] Clearing atomic state 93ad4380
[  303.434979] [drm:__drm_atomic_state_free] Freeing atomic state 93ad4380
[  305.383433] ftgmac100 1e660000.ethernet eth0: NCSI: LSC AEN - channel 0 state up
[  305.451550] ftgmac100 1e660000.ethernet eth0: NCSI: host driver running on channel 0
[  318.051678] ftgmac100 1e660000.ethernet eth0: NCSI: host driver not running on channel 0
[  347.853883] ftgmac100 1e660000.ethernet eth0: NCSI: host driver running on channel 0
root@witherspoon:~#
```

rfrandse, 2018-06-28:
<https://gerrit.openbmc-project.xyz/11319> kernel: FSI, ASPEED SPI NOR, ASPEED GFX
Resolves:  #3164 Withespoon Kernel CRTC: vblank wait timed out

closed 2018-07-04 by geissonator