openbmc/openbmc #3176: Fan Watchdog: Create dump when fan watchdog trips

msbarth, 2018-05-10:
The fan watchdog should not be expected to trip on purpose and because of this, a dump should be created containing the current state of the BMC at the time the fan watchdog trips.
This can be handled by a service that runs `dreport` once the fan watchdog target has been set which conflicts with the service that keeps the watchdog from tripping.

rfrandse, 2018-05-14:
<https://gerrit.openbmc-project.xyz/10593> wspoon: Create dump when fan watchdog trips
Resolves:  #3176 Fan Watchdog: Create dump when fan watchdog trips

msbarth, 2019-04-04:
Pending rework of https://gerrit.openbmc-project.xyz/10593 to what was merged under https://github.com/ibm-openbmc/dev/issues/150 on 1/30

amboar, 2019-04-08:
Chances are the watchdog is going to bite because of some load-related issue on the BMC. Have you considered impact of the added load of triggering a dump?

msbarth, 2019-04-08:
@amboar That is actually part of the intention for having this dump collection. When the fan watchdog trips, for whatever reason it was, we need to collect a dump to get enough data to determine what/why the fan watchdog tripped since the watchdog can only be reset by reboot the BMC.

Do you have a better suggestion that wouldn't add a heavy load like a dump but collect the same amount of data that a dump does?

amboar, 2019-04-09:
> Do you have a better suggestion that wouldn't add a heavy load like a dump but collect the same amount of data that a dump does?

Haha, no, other than reproducing the issue and live debugging the system. You're observing this problem in the field?

amboar, 2019-04-10:
Right, sounds like a reasonable trade-off. Just wary of it causing cascading failures due to timeouts from CPU starvation.

spinler, 2019-04-10:
Just to put this out there- no code actually knows when the hardware watchdog trips.  That problem we had awhile ago was when the watchdog app couldn't ping fast enough, so the watchdog did indeed trip but that app didn't even know it.

We would only capture a dump when we can tell the watchdog trips, which is when a fan process dies so the watchdog app gets killed (due to how we set up service dependencies) so the hw watchdog expires.

msbarth, 2019-04-10:
When I'm able to get back to working this, I'll poke around to see if there'd be a way I could code something up so we could tell the fan watchdog status. It'd be beneficial if we can have the ability to also collect a dump when it may have tripped due to something no specifically configured to trip it (ie the fan processes).

msbarth, 2019-10-11:
unstale

closed 2020-05-12 by stale[bot]