openbmc/openbmc #3178: Server configuration -> Firmware -> Specify image file location Issue

cdeadmin, 2018-05-11:


anoo1, 2018-05-11:
User reports:
```
BMC GUI: Server configuration -> Firmware -> Specify image file location

Issue: While trying to upload BMC image OR Server image, getting the error as "Error uploading image." But image is seen as listed by refreshing GUI after some time.
```
Andrew is requesting nginx extended data:
```
#ssh to bmc and change nginx to be more verbose
vi /etc/nginx/nginx.conf
Put a "#" in front of "access_log  off;" and save change
systemctl restart nginx.service
```
Closing for now until data is collected.

geissonator, 2018-05-15:
Testing in certain network environments has show the 10s
timeout is too small for nginx. It's unclear if the issue is with the
client or server side. When it hits, it occurs during the
upload of BMC and PNOR images.

The error on the client side is a errno 104 (Connection reset
by peer). The error on the server side is a 408 which in nginx
indicates a timeout for the client_body_timeout setting or the
client_header_timeout setting. It's doubtful the client_header
could be the issue here so only the client_body_timeout was
focused on.

The send_timeout value can also result in a dropped client connection.

The default for both the client_body_timeout and the send_timeout is
60s. In openbmc these were changed to 10s to match with OWASP
guidelines. 

https://gerrit.openbmc-project.xyz/#/c/10613/ tracks the increase in timeout.

geissonator, 2018-07-09:
Yeah, we just needed the additional time, I don't think there's any underlying fix to make it faster.

closed 2018-07-09 by geissonator