openbmc/openbmc #3181: OP920: Errorlog rotation failed

cdeadmin, 2018-05-14:


spinler, 2018-05-14:
There is a case where another service (ibm-logging) provides additional interfaces on the phosphor-logging entry objects.  This service has its object lifespans exactly match the phosphor-logging ones already by listening for the interfaces-added and intefaces-removed signals, and so it has no use for a Delete interface.

The current REST code would return a 403 on a DELETE if any service with the specified path didn't provide a Delete interface.  The fix will just skip over objects that don't have that interface, and only return a 403 error if no Delete calls at all were made.

rfrandse, 2018-05-14:
<https://gerrit.openbmc-project.xyz/10597> Don't fail on missing Delete interfaces
Resolves  #3181  Errorlog rotation failed

closed 2018-05-29 by geissonator