openbmc/openbmc #3186: Tighten ssh encryption protocols and web server xss protection

joseph-reynolds, 2018-05-16:
SW429156
This is in response to a security scan.  There are four parts:
1. Disable ssh triple-DES ("DES-CBC3").
2. Disable ssh CBC mode cipher encryption, and enable CTR or GCM cipher mode encryption.
3. Disable ssh MD5 and 96-bit MAC algorithms.
4. Update the web server to protect from XSS vulnerability.

joseph-reynolds, 2018-05-17:
Maybe we should go to a more recent openssh version?  It looks like openbmc is using OpenSSH version **7.5p1** (openbmc/import-layers/yocto-poky/meta/recipes-connectivity/openssh/openssh_7.5p1.bb) and **7.7p1** is available.

If we upgrade, are there any concerns connecting older ssh clients?  For example RHEL 6.8 uses OpenSSH 5.3p1 (I think) which ought to able to connect to an OpenSSH 7.7 server (I think).

According to https://www.openssh.com/releasenotes.html:
OpenSSH 7.6 disables CBC by default
OpenSSH 5.1 prefers CTR
OpenSSH 3.7.1 added support for CTR
OpenSSH 6.2 added support for GCM
OpenSSH 7.2 disables MD5
OpenSSH 6.1 removed hmac-sha2-256-96 and hmac-sha2-512-96

joseph-reynolds, 2018-05-18:
Going to OpenSSH 7.6 would be good enough because it disables CBC.  We wouldn't need to go to 7.7.

If we must stay at OpenSSH 7.5p1 and cannot go to 7.6, we could add a /etc/sshd/sshd-config file that disables CBC.

joseph-reynolds, 2018-05-18:
According to https://www.yoctoproject.org/software-overview/downloads/,
release information for "YP Core - Sumo 2.5"
contains this entry: openssh: update to 7.6
So, updating to yocto poky 2.5 will fix the openssh issue.

joseph-reynolds, 2018-05-29:
Oops.  Although Yocto includes both OpenSSH and Dropbear SSH, I was informed our implementation uses dropbear.  (So fixes to OpenSSH won't fix this problem.)  Issue #1229 has more detail for dropbear.

joseph-reynolds, 2018-08-06:
To configure dropbear to drop the insecure ciphers, according to https://github.com/mkj/dropbear/blob/master/default_options.h, we need to create a new file:
```
localoptions.h
#define DROPBEAR_TWOFISH256 0
#define DROPBEAR_TWOFISH128 0
#define DROPBEAR_ENABLE_CBC_MODE 0
#define DROPBEAR_SHA1_HMAC 0
#define DROPBEAR_SHA1_96_HMAC 0
```

joseph-reynolds, 2018-08-09:
This has details about how to update Dropbear's configuration.

To control Dropbear SSH server features (including supported ciphers) you must update a header file and recompile.  Dropbear changed its config mechanism between its 2017.75 and 2018.76 releases:
 - For 2017.75 and earlier, you patch the options.h file.
 - For 2018.76 and later, you provide a new localoptions.h file which is used if present and "overrides" C++ preprocessor macros in default_options.h.

We are currently using 2017.75.  Here are details for upgrading to 2018.76:
 1. Delete file recipes-core/dropbear/dropbear/options.patch.
 2. Create new file recipes-core/dropbear/dropbear/localoptions.h with:
```
/* Customize dropbear per default_options.h in the dropbear project */

/* Disable insecure ciphers */
#define DROPBEAR_TWOFISH256 0
#define DROPBEAR_TWOFISH128 0
#define DROPBEAR_ENABLE_CBC_MODE 0
#define DROPBEAR_SHA1_HMAC 0
#define DROPBEAR_SHA1_96_HMAC 0
```
 3. Update file recipes-core/dropbear/dropbear.inc as follows:
```
SRC_URI += "file://localoptions.h"
do_unpack_extra() {
    install -m 0644 ${WORKDIR}/localoptions.h ${S}
}
addtask unpack_extra after do_unpack before do_patch
```
4. These changes have not been tested, so ... test away! 

Notes:
 - The dropbear recipe creates new bitbake task do_unpack_extra because do_unpack is Python and do_unpack_extra has shell commands.
 - The Dropbear build picks up the localoptions.h file automatically via its Makefile.
 - To see which ciphers Dropbear supports, try: `ssh -c help 127.0.0.1`.
 - To connect from the QEMU host, try: `ssh -c aes128-ctr -p 2222 -l root localhost`.

joseph-reynolds, 2018-08-09:
I created the dropbear patch like this:
```
 git clone https://github.com/mkj/dropbear.git
 cd dropbear
 git checkout beaff53a79a98317650c47fd821be1075e4369fe
 ...edit options.h to remove the unwanted defines
 git diff > options.patch
```
The git commit-id above was the last update to options.h, not the last commit before the release.

rfrandse, 2018-08-13:
<https://gerrit.openbmc-project.xyz/11894> Disable medium-strength dropbear ssh ciphers
Resolves  #3186 Tighten ssh encryption protocols and web server xss protection

closed 2018-09-27 by geissonator

joseph-reynolds, 2018-10-26:
The above review was abandoned, moved to a different repo, and merged here:
  https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/13278/
That patch is being changed to allow HMAC_SHA1 here:
  https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/13891/
My attempt to upstream to the dropbear project is pending.

joseph-reynolds, 2019-04-15:
This is being updated for OpenBMC 2.7 (as master branch) here: https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/20516