openbmc/openbmc #3187: Support averaging OCC power values in phosphor-hwmon

spinler, 2018-05-16:
The OCC reports instantaneous power readings in its power*_input values.  The GPU values especially appear to be all over the place, and we were asked to report average values instead for the GPU power sensors, like the nvidia-smi command line tools does.

Can do this by reading the power*_update_tag and power*_accumulator hwmon files, and doing:
```
average between poll1 and poll2 = (accumulator2 - accumulator1)/(updateTag2 - updateTag1)
```

where:

update_tag: 
```
 4 bytes. Count of number of 250us samples represented in Accumulator. Used for time derived sensor.
```
accumulator:
```
 8 bytes. Accumulation of 250us power readings
```

mdmillerii, 2018-05-20:
@spinler  please expand with how you will make sure an update doesn't occur between reading the `accumulator` and the `update_tag` files.  Is there an interlock in the driver needed?  Do you need to read the tag before and after, and retry if different?

spinler, 2018-05-21:
@mdmillerii  We haven't worked through everything yet, but will keep that in mind when we do.  This item isn't scheduled yet.

rfrandse, 2019-07-11:
<https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-hwmon/+/23449> Support averaging OCC power values in phosphor-hwmon
Resolves:  #3187 Support averaging OCC power values in phosphor-hwmon

closed 2019-08-30 by geissonator