openbmc/openbmc #3195: Tighten web server xss protection

joseph-reynolds, 2018-05-18:
This was split off #3186 to isolate the web server XSS issue from the ssh issues.
SW429156
This is in response to a security scan:
 - Update the web server to protect from XSS vulnerability.

joseph-reynolds, 2018-05-25:
This appears to be a rediscovery of #1229.

joseph-reynolds, 2018-08-01:
See also #3328.

joseph-reynolds, 2018-08-01:
The phosphor-rest-server had its XSS vulnerability addressed by https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-rest-server/+/8483/ by @causten by adding HTTP headers:
```
            'X-Frame-Options', 'deny'
            'X-Content-Type-Options', 'nosniff'
            'X-XSS-Protection', '1; mode=block'
            'Content-Security-Policy', "default-src 'self'"
            'Strict-Transport-Security', 'max-age=31536000; includeSubDomains; preload'
```

Can we fix the remaining problems by changing nginx, phosphor-webui, or both?

The configuration changes for nginx would be something like:
```
	add_header "X-XSS-Protection" "1; mode=block";
	add_header "Content-Security-Policy" "default-src 'self'";
	etc.
```

The changes to phosphor-webui (NodeJS?) would be to add HTTP headers as above.

For a little light reading:
- https://developer.mozilla.org/en-US/docs/Web/HTTP/CSP
- https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/X-XSS-Protection

joseph-reynolds, 2018-08-01:
The phosphor-rest-server had its XSS vulnerability addressed by https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-rest-server/+/8483/ by adding HTTP headers:
```
            'X-Frame-Options', 'deny'
            'X-Content-Type-Options', 'nosniff'
            'X-XSS-Protection', '1; mode=block'
            'Content-Security-Policy', "default-src 'self'"
            'Strict-Transport-Security', 'max-age=31536000; includeSubDomains; preload'
```

Can we fix the remaining problems by changing nginx, phosphor-webui, or both?

The configuration changes for nginx would be something like:
```
	add_header "X-XSS-Protection" "1; mode=block";
	add_header "Content-Security-Policy" "default-src 'self'";
	etc.
```

The changes to phosphor-webui (NodeJS?) would be to add HTTP headers as above.

For a little light reading:
- https://developer.mozilla.org/en-US/docs/Web/HTTP/CSP
- https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/X-XSS-Protection

joseph-reynolds, 2018-08-01:
See also closed issue #2423.

joseph-reynolds, 2018-09-07:
Consider using the same values are bmcweb: https://github.com/openbmc/bmcweb/blob/master/include/security_headers_middleware.hpp

joseph-reynolds, 2018-09-11:
Using
> Content-Security-Policy: default-src 'self'
causes the web application to not load properly.  The web app uses Angular, so to make it load without errors I changed the HTTP headers like this:
> Content-Security-Policy: default-src 'self' 'unsafe-eval' 'unsafe-inline'
But we (@gunnar?) should be following the recommendations in https://docs.angularjs.org/api/ng/directive/ngCsp.

rfrandse, 2018-09-20:
<https://gerrit.openbmc-project.xyz/12247> Nginx adds http security headers
Resolves  #3195 Tighten web server xss protection

closed 2018-09-24 by bradbishop