openbmc/openbmc #3197: Cannot get IPv6 from DHCP server and configure question

KenChenIEC, 2018-05-21:
Our system can get IPv4 from DHCP but IPv6 doesn't. 
Could someone tell me is it work as design?
Because I found the config file is setting the DHCP on IPv4.
https://github.com/openbmc/openbmc/blob/171031d20c7ed03900739d51ba53ad0001f98fa5/meta-phosphor/common/recipes-core/systemd/systemd/default.network

BTW, if I need to enable the ipv6, I just need to add the DHCP=ipv4, ipv6?

Thanks.

dcobbley, 2018-05-24:
default.network is a backup network file which is not used if /etc/systemd/network/eth<deviceNumber>.network exists.

The phosphor-networkd writes the file in /etc upon initial boot up. In there you should see DHCP=true.

I do not have access to an IPv6 DHCP server, but do have the ability to give myself an IPv6 address and works fine on my platform.

KenChenIEC, 2018-06-01:
the /etc/systemd/network file as below

> root@lanyang:/etc/systemd/network# ls
> 00-bmc-eth0.network  00-bmc-eth1.network
> root@lanyang:/etc/systemd/network# cat 00-bmc-eth1.network
> [Match]
> Name=eth1
> [Network]
> DHCP=true
> LinkLocalAddressing=yes
> IPv6AcceptRA=false
> [DHCP]
> ClientIdentifier=mac

Is the IPv6AcceptRA=false reason of not get DHCP?
If yes, how can I modify this?

My platform also has IPv6 by self-generate.

> 3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
>     link/ether 6a:e1:03:b3:8e:53 brd ff:ff:ff:ff:ff:ff
>     inet 169.254.249.3/16 brd 169.254.255.255 scope link eth1
>        valid_lft forever preferred_lft forever
>     inet 10.6.45.56/24 brd 10.6.45.255 scope global dynamic eth1
>        valid_lft 249756sec preferred_lft 249756sec
>     inet6 fe80::68e1:3ff:feb3:8e53/64 scope link
>        valid_lft forever preferred_lft forever

bluecmd, 2018-10-20:
I was having trouble getting NC-SI on Linux to join the `ff02::1` multicast group that RAs are being sent to, I'm not sure the same issue is present in OpenBMC, but if it is that might be the issue here.

closed 2019-10-15 by stale[bot]