openbmc/openbmc #3202: [LTCTest][OPAL][OP920] ipmid[1224]: MgfDate invalid date: 820454400 secs since UNIX epoch

cdeadmin, 2018-05-22:


gtmills, 2018-05-30:
@tomjoseph83 Can you add details about this?

tomjoseph83, 2018-05-31:
When `ipmitool fru list` command is executed, the following traces is added to the journal. The trace is added for both good path/bad path. The trace is needed only for the bad path case.

```
May 17 05:25:09 witherspoon phosphor-timemanager[1326]: BMC system time is changed
May 17 05:27:39 witherspoon ipmid[1224]: ERROR:[0xCB] handling NetFn:[0xA], Cmd:[0x10]
May 17 05:27:42 witherspoon ipmid[1224]: MgfDate invalid date: 820454400 secs since UNIX epoch
May 17 05:27:42 witherspoon ipmid[1224]: MgfDate invalid date: 820454400 secs since UNIX epoch
May 17 05:27:42 witherspoon ipmid[1224]: MgfDate invalid date: 820454400 secs since UNIX epoch
May 17 05:27:45 witherspoon ipmid[1224]: MgfDate invalid date: 4294967295 secs since UNIX epoch
May 17 05:27:45 witherspoon ipmid[1224]: MgfDate invalid date: 4294967295 secs since UNIX epoch
May 17 05:27:45 witherspoon ipmid[1224]: MgfDate invalid date: 4294967295 secs since UNIX epoch
```

rfrandse, 2018-07-11:
<https://gerrit.openbmc-project.xyz/11426> Fix MgfDate invalid date: 820454400 secs since UNIX epoch
Resolves  #3202 [OPAL] ipmid[1224]: MgfDate invalid date: 820454400 secs since UNIX epoch

closed 2018-07-31 by geissonator