openbmc/openbmc #3205: IPL: Power is off after power on request on w50 with 910.1806.20180522a

cdeadmin, 2018-05-25:


spinler, 2018-05-25:
On Witherspoon during a power on, we have to unbind the 4 VRM device drivers, run some scripts to do some workarounds, and then rebind them.  We had an instance where the probe that happens in a bind failed with an EBADMSG errno.  Then, because the bind service is a required by the power on target, the power on failed.

Will do 2 things to make this more resilient:

1. Retry the bind a few times if it fails.  The UCD90160 workaround scripts already do this.
2. Move the service to a wants dependency instead of requires, so if it still fails the power on will still continue and we may just not get the hwmon readings for those VRMs.  If it happens to be on a water cooled system it may make fan control mad a little bit, but it should be able to handle it and I think it's still better than not powering on at all.

rfrandse, 2018-05-25:
<https://gerrit.openbmc-project.xyz/10846> wspoon: Change ir35221-on-bind.service dependency
Resolves  #3205 Witherspoon:  VRM bind failed on a power on and caused the power on to fail

closed 2018-05-30 by geissonator