openbmc/openbmc #3206: Build fails with `No rule to make target 'arch/arm/boot/dts/aspeed-bmc-opp-firestone.dtb`

paulmenzel, 2018-05-25:
```
$ git describe
v2.2-182-g5253b55a
$ git log --oneline -1
5253b55a (HEAD -> master, origin/master, origin/HEAD) npcm750-evb: added support for npcm750-evb
$ TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-firestone/conf . oe-init-build-env
$ bitbake obmc-phosphor-image
[…]
|   Kernel: arch/arm/boot/Image is ready
|   Kernel: arch/arm/boot/zImage is ready
| NOTE: make -j 160 HOSTCC=gcc  -isystem/home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/recipe-sysroot-native/usr/lib -L/home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/recipe-sysroot-native/lib -Wl,-rpath-link,/home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/recipe-sysroot-native/lib -Wl,-O1 HOSTCPP=gcc  -E aspeed-bmc-opp-firestone.dtb
|   CC      scripts/mod/empty.o
|   CC      scripts/mod/devicetable-offsets.s
|   MKELF   scripts/mod/elfconfig.h
|   HOSTCC  scripts/mod/modpost.o
|   HOSTCC  scripts/mod/sumversion.o
|   CHK     scripts/mod/devicetable-offsets.h
|   HOSTCC  scripts/mod/file2alias.o
|   HOSTLD  scripts/mod/modpost
| make[3]: *** No rule to make target 'arch/arm/boot/dts/aspeed-bmc-opp-firestone.dtb'.  Stop.
| arch/arm/Makefile:342: recipe for target 'aspeed-bmc-opp-firestone.dtb' failed
| make[2]: *** [aspeed-bmc-opp-firestone.dtb] Error 2
| Makefile:145: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:24: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/joey/src/openbmc/build/tmp/work/firestone-openbmc-linux-gnueabi/linux-obmc/4.13.16+gitAUTOINC+28d817669a-r0/temp/log.do_compile.38958)
```

mine260309, 2018-05-28:
According to https://github.com/openbmc/openbmc/tree/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-firestone:
> Firestone is no longer a supported machine.

Firestone's device tree is removed from kernel, so it fails the build.

paulmenzel, 2018-05-28:
Thank you. I didn’t know that. But as it’s in the tree, I would at least expect it to build.

> Firestone's device tree is removed from kernel, so it fails the build.

Was it ever in upstream Linux kernel or just in OpenBMC? I couldn’t find it in the upstream Linux kernel. But, it seems to be in the [OpenBMC Linux kernel](https://github.com/openbmc/linux/blob/dev-4.10/arch/arm/boot/dts/aspeed-bmc-opp-firestone.dts).

@gtmills, as you added the “unsupported comment”, could you please elaborate? We bought the system last year and there is a two year warranty on it. So I would have expected it to be supported with security updates at least.

paulmenzel, 2018-05-29:
It’s only in the branch *dev-4.10* and not in *dev-4.13*. How confusing.

bradbishop, 2018-05-29:
@paulmenzel Sorry about the confusion but OpenBMC on Firestone systems is not supported by IBM or anyone.  Firestones do not come from the factory with OpenBMC installed and there isn't any way to install OpenBMC on a Firestone without voiding the warranty.

> But as it’s in the tree, I would at least expect it to build.

This is a completely normal expectation.  We'll have someone remove it from the tree.

paulmenzel, 2018-05-29:
On 05/29/18 17:10, Brad Bishop wrote:
> @paulmenzel Sorry about the confusion but OpenBMC on Firestone
> systems is not supported by IBM or anyone.  Firestones do not come
> from the factory with OpenBMC installed and there isn't any way to
> install OpenBMC on a Firestone without voiding the warranty.

Oh, I didn’t know that. I hoped as OpenPOWER system, it would have 
OpenBMC. Do you know what it has, and if I can be updated? I need a 
newer SSH server in there. ;-)

>> But as it’s in the tree, I would at least expect it to build.
> 
> This is a completely normal expectation.  We'll have someone remove
> it from the tree.

Depending on how difficult it would be, I’d be willing to work on 
getting OpenBMC on Firestone systems.

paulmenzel, 2018-05-29:
For the record, here is the system information.

```
fru print 47
 Product Name          : OpenPOWER Firmware
 Product Version       : IBM-firestone-ibm-OP8_v1.12_2.85
 Product Extra         : 	op-build-d033e11
 Product Extra         : 	buildroot-81b8d98
 Product Extra         : 	skiboot-5.4.8-7f3e3b0
 Product Extra         : 	hostboot-2eb7706-f28ad92
 Product Extra         : 	linux-4.4.92-openpower1-59284a2
 Product Extra         : 	petitboot-v1.4.4-a6d3938
 Product Extra         : 	firestone-xml-2494a43
 Product Extra         : 	occ-d7efe30-dca9
```

amboar, 2018-06-01:
> Oh, I didn’t know that. I hoped as OpenPOWER system, it would have OpenBMC.

Ah, Firestone was a victim of timing more than anything. IBM began its OpenBMC journey after the OpenPOWER ecosystem was pumping out machines, so there are a number of OpenPOWER platforms that use AMI BMC firmware.

Glad to hear your enthusiasm for opening up the firmware stack though :)

shenki, 2018-06-01:
Firestone is not an OpenBMC platform. The machines ship with another vendor's BMC firmware.

One of the hurdles you would need to overcome is the BMC flash chip: it ships with a 16MB NOR, but OpenBMC requires at least 32MB.

FYI, I have posted a patch to remove it from the tree: https://gerrit.openbmc-project.xyz/c/10957/

gtmills, 2018-06-29:
@paulmenzel Firestone is removed. Can we close this issue?

closed 2018-08-20 by gtmills