openbmc/openbmc #3207: Host incorrectly powers off straight away

shenki, 2018-05-29:
When using the op-test suite to perform boot testing, I noticed openbmc queues up a poweroff request when the machine is already powered off. If the host then powers on before that poweroff request expires (30 seconds), it causes the machine to shut down.

### Normal behaviour
Check machine state is off:
```
$ curl -s  -b cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : []}'  -X GET https://ozrom1-bmc///xyz/openbmc_project/state/host0
{
  "data": {
    "AttemptsLeft": 3, 
    "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified", 
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off", 
    "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.Inactive", 
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```
Tell the BMC to perform an off transition:
```
$ curl c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.Off"}' -X PUT https://${bmc}/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}
```

On the BMC, we see it tries to power off:
```
May 29 06:43:49 ozrom1-bmc phosphor-host-state-manager[1043]: Host State transaction request
May 29 06:43:49 ozrom1-bmc systemd[1]: Stopped target Chassis0 (Reset Check).
May 29 06:43:50 ozrom1-bmc systemd[1]: Stopped target Chassis0 power on after reset.
May 29 06:43:50 ozrom1-bmc systemd[1]: Starting OpenPOWER OCC Active Disable...
May 29 06:43:50 ozrom1-bmc systemd[1]: Started Power is off to chassis0.
May 29 06:43:50 ozrom1-bmc systemd[1]: Starting Soft power off of the host...
May 29 06:43:50 ozrom1-bmc ipmid[974]: Pushing cmd on to queue
May 29 06:43:50 ozrom1-bmc ipmid[974]: Pushing cmd on to queue
May 29 06:43:50 ozrom1-bmc ipmid[974]: Asserting SMS Attention
May 29 06:43:50 ozrom1-bmc ipmid[974]: SMS Attention asserted
May 29 06:43:50 ozrom1-bmc systemd[1]: Started Soft power off of the host.
May 29 06:43:50 ozrom1-bmc phosphor-mapper[893]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff discovery failure on / - retry
May 29 06:43:50 ozrom1-bmc systemd[1]: Started OpenPOWER OCC Active Disable.
May 29 06:44:20 ozrom1-bmc ipmid[974]: Timer expired
May 29 06:44:20 ozrom1-bmc ipmid[974]: Host control timeout hit!
```

Check state is off:
```
$ curl s  -b cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : []}'  -X GET https://ozrom1-bmc///xyz/openbmc_project/state/host0
{
  "data": {
    "AttemptsLeft": 3, 
    "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified", 
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off", 
    "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.Inactive", 
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
````

```
$ curl c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.On"}' -X PUT https://${bmc}/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}
```

The machine boots as expected.

### Buggy behaviour
If we perform the off, status check, on steps back to back:
```
$ curl c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.Off"}' -X PUT https://${bmc}/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}
$  curl s  -b cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : []}'  -X GET https://ozrom1-bmc///xyz/openbmc_project/state/host0
{
  "data": {
    "AttemptsLeft": 3, 
    "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified", 
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off", 
    "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.Inactive", 
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
$  curl c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.On"}' -X PUT https://${bmc}/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}
```

The system comes up, sees the pending power off request, and shuts down:
```
--== Welcome to Hostboot  ==--

  2.61552|secure|SecureROM valid - enabling functionality
  3.72759|IPMI: shutdown requested
  3.82136|Booting from SBE side 0 on master proc=00050000
  4.44449|IPMI: shutdown complete
```

shenki, 2018-05-29:
I could reproduce on Romulus with 2.2-22 and v2.2-187-gf0ecbe4 (master as of today)

stewart-ibm, 2018-05-29:
Ouch. This is a pretty major regression :(

We could change `op-test` to check the state of the machine before hammering it I guess, but this is clearly an API regression on the OpenBMC side.

geissonator, 2018-06-01:
It seems like the best option is to be sure to clear the BMC host communication queue when the host poweron target is started.

spinler, 2018-06-27:
I made and tested the fix suggested above.  The review is at https://gerrit.openbmc-project.xyz/#/c/11305/.

rfrandse, 2018-06-27:
<https://gerrit.openbmc-project.xyz/11305> Clear host command queue on a power on
Resolves  #3207 Host incorrectly powers off straight away

closed 2018-07-24 by geissonator