openbmc/openbmc #3208: OpenBMC build errors with Ubuntu 16.04.4 and Virtualbox

Aeneas32, 2018-05-29:
I continue to receive errors when I try an initial build of OpenBMC
under  //Windows 7 64//Virtualbox//Ubuntu 16.04.4// .
1) Can OpenBMC be built under Ubuntu 16.04, (instead of Ubuntu 14.04) and
2) can OpenBMC be built under Virtualbox ?

Aeneas32, 2018-05-29:
Ok:
=========================================================
=========================================================
aen@aen-VirtualBox:~$ sudo apt-get install -y git build-essential libsdl1.2-dev texinfo gawk chrpath diffstat
[sudo] password for aen: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
diffstat is already the newest version (1.61-1).
gawk is already the newest version (1:4.1.3+dfsg-0.1).
libsdl1.2-dev is already the newest version (1.2.15+dfsg1-3).
chrpath is already the newest version (0.16-1).
texinfo is already the newest version (6.1.0.dfsg.1-5).
git is already the newest version (1:2.7.4-0ubuntu1.3).
The following packages were automatically installed and are no longer required:
  linux-headers-4.13.0-36 linux-headers-4.13.0-36-generic linux-image-4.13.0-36-generic linux-image-extra-4.13.0-36-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 31 not upgraded.
aen@aen-VirtualBox:~$ git clone git@github.com:openbmc/openbmc.git
Cloning into 'openbmc'...
remote: Counting objects: 61600, done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 61600 (delta 30), reused 18 (delta 18), pack-reused 61545
Receiving objects: 100% (61600/61600), 31.93 MiB | 5.66 MiB/s, done.
Resolving deltas: 100% (29430/29430), done.
Checking connectivity... done.
Checking out files: 100% (10698/10698), done.
aen@aen-VirtualBox:~$ ls
Desktop  Documents  Downloads  examples.desktop  Music  openbmc  Pictures  Public  Templates  Videos
aen@aen-VirtualBox:~$ cd openbmc
aen@aen-VirtualBox:~/openbmc$ 
aen@aen-VirtualBox:~/openbmc$ gedit README.md
aen@aen-VirtualBox:~/openbmc$ printenv TEMPLATECONF
meta-openbmc-machines/meta-openpower/meta-ibm/meta-palmetto/conf

aen@aen-VirtualBox:~/openbmc/build$ export TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-palmetto/conf
aen@aen-VirtualBox:~/openbmc/build$ cd ..
aen@aen-VirtualBox:~/openbmc$ . openbmc-env
### Initializing OE build env ###
Common targets are:
     obmc-phosphor-image
aen@aen-VirtualBox:~/openbmc/build$ bitbake obmc-phosphor-image
Parsing recipes: 100% |###############################################################################################################################| Time: 0:17:48
Parsing of 2240 .bb files complete (0 cached, 2240 parsed). 3108 targets, 319 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "palmetto"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm armv5 thumb dsp"
TARGET_FPU           = "soft"
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-perl            
meta-python          
meta-webserver       
meta-virtualization  
meta-phosphor        
meta-aspeed          
meta-ast2400         
meta-openpower       
meta-ibm             
[... 11 lines cut ...]
ERROR: Logfile of failure stored in: /home/aen/openbmc/build/tmp/work/palmetto-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.10993
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/aen/openbmc/meta/recipes-devtools/quilt/quilt-native_0.65.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-bsp/u-boot/u-boot-mkimage_2017.09.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-devtools/binutils/binutils-cross_2.29.1.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-extended/xz/xz_5.2.3.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-extended/bc/bc_1.06.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.5.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-devtools/createrepo-c/createrepo-c_git.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-devtools/gcc/gcc-cross_7.3.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-devtools/squashfs-tools/squashfs-tools_git.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-devtools/dnf/dnf_2.6.3.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot', '/home/aen/openbmc/meta-openbmc-bsp/meta-aspeed/common/recipes-bsp/u-boot/u-boot_2016.07.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-core/os-release/os-release.bb:do_populate_sysroot', '/home/aen/openbmc/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-devtools/opkg/opkg_0.3.5.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-core/update-rc.d/update-rc.d_0.7.bb:do_populate_sysroot', 'virtual:native:/home/aen/openbmc/meta/recipes-devtools/mtd/mtd-utils_gi [... cut ...]
NOTE: Tasks Summary: Attempted 3998 tasks of which 18 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/aen/openbmc/meta-phosphor/common/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static
Summary: There were 5 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
aen@aen-VirtualBox:~/openbmc/build$ 
=============================================================
=============================================================

mdmillerii, 2018-05-29:
@Aeneas32 Github issues use markdown.  To prevent ascii text characters from being interpreted as formatting you need to escape your log content with ` ``` ` on a line before and after your log.

You can find more information here: https://guides.github.com/features/mastering-markdown/ . 

(You can use the edit button to adjust the log above to make it easier for all to read).
`

mdmillerii, 2018-05-29:
@Aeneas32 That said, it appears that a late stage packaging function is saying a file does not exist.  Was this a clean build from an empty directory or had you restarted along the way?  `bitbake` maintains state files recording what has already completed and will not attempt to generate intermediate output if the stamp files say the file is there. 

Can you check what part of this path exists `/home/aen/openbmc/build/tmp/deploy/images/palmetto/u-boot.bin` ?

Aeneas32, 2018-05-29:
I deleted the openbmc directory then git clone-ed again.
Is there a  make clean  equivalent in this yocto build ?
```
aen@aen-VirtualBox:~/openbmc/build/tmp/deploy/images$ cd palmetto
aen@aen-VirtualBox:~/openbmc/build/tmp/deploy/images/palmetto$ ls -l
total 13236
lrwxrwxrwx 2 aen aen      77 May 23 15:23 aspeed-bmc-opp-palmetto.dtb -> zImage--4.13.16+git0+468cbec6d2-r0-aspeed-bmc-opp-palmetto-20180523090537.dtb
lrwxrwxrwx 2 aen aen      64 May 23 15:23 fitImage -> fitImage--4.13.16+git0+468cbec6d2-r0-palmetto-20180523090537.bin
-rw-r--r-- 2 aen aen 2356420 May 23 15:23 fitImage--4.13.16+git0+468cbec6d2-r0-palmetto-20180523090537.bin
-rw-r--r-- 2 aen aen    1580 May 23 15:23 fitImage-its-4.13.16+gitAUTOINC+468cbec6d2-r0-palmetto-20180523090537.its
-rw-r--r-- 2 aen aen    2183 May 23 15:23 fitImage-its-obmc-phosphor-initramfs-palmetto-4.13.16+gitAUTOINC+468cbec6d2-r0-20180523090537.its
lrwxrwxrwx 2 aen aen      97 May 23 15:23 fitImage-its-obmc-phosphor-initramfs-palmetto.its -> fitImage-its-obmc-phosphor-initramfs-palmetto-4.13.16+gitAUTOINC+468cbec6d2-r0-20180523090537.its
lrwxrwxrwx 2 aen aen      73 May 23 15:23 fitImage-its-palmetto.its -> fitImage-its-4.13.16+gitAUTOINC+468cbec6d2-r0-palmetto-20180523090537.its
-rw-r--r-- 2 aen aen 2329848 May 23 15:23 fitImage-linux.bin-4.13.16+gitAUTOINC+468cbec6d2-r0-palmetto-20180523090537.bin
lrwxrwxrwx 2 aen aen      79 May 23 15:23 fitImage-linux.bin-palmetto.bin -> fitImage-linux.bin-4.13.16+gitAUTOINC+468cbec6d2-r0-palmetto-20180523090537.bin
-rw-r--r-- 2 aen aen 4224236 May 23 15:23 fitImage-obmc-phosphor-initramfs-palmetto-4.13.16+gitAUTOINC+468cbec6d2-r0-20180523090537.bin
lrwxrwxrwx 2 aen aen      93 May 23 15:23 fitImage-obmc-phosphor-initramfs-palmetto.bin -> fitImage-obmc-phosphor-initramfs-palmetto-4.13.16+gitAUTOINC+468cbec6d2-r0-20180523090537.bin
lrwxrwxrwx 2 aen aen      64 May 23 15:23 fitImage-palmetto.bin -> fitImage--4.13.16+git0+468cbec6d2-r0-palmetto-20180523090537.bin
-rw-r--r-- 1 aen aen  110054 May 23 17:21 obmc-phosphor-debug-tarball-palmetto-20180523090537.testdata.json
lrwxrwxrwx 1 aen aen      65 May 23 17:21 obmc-phosphor-debug-tarball-palmetto.testdata.json -> obmc-phosphor-debug-tarball-palmetto-20180523090537.testdata.json
-rw-r--r-- 1 aen aen  116849 May 23 17:03 obmc-phosphor-image-palmetto-20180523090537.testdata.json
lrwxrwxrwx 1 aen aen      57 May 23 17:03 obmc-phosphor-image-palmetto.testdata.json -> obmc-phosphor-image-palmetto-20180523090537.testdata.json
-rw-r--r-- 2 aen aen 1867594 May 23 14:49 obmc-phosphor-initramfs-palmetto-20180523090537.rootfs.cpio.lzma
-rw-r--r-- 2 aen aen     985 May 23 13:59 obmc-phosphor-initramfs-palmetto-20180523090537.rootfs.manifest
-rw-r--r-- 1 aen aen  112301 May 23 13:59 obmc-phosphor-initramfs-palmetto-20180523090537.testdata.json
lrwxrwxrwx 2 aen aen      64 May 23 14:49 obmc-phosphor-initramfs-palmetto.cpio.lzma -> obmc-phosphor-initramfs-palmetto-20180523090537.rootfs.cpio.lzma
lrwxrwxrwx 2 aen aen      63 May 23 13:59 obmc-phosphor-initramfs-palmetto.manifest -> obmc-phosphor-initramfs-palmetto-20180523090537.rootfs.manifest
lrwxrwxrwx 1 aen aen      61 May 23 13:59 obmc-phosphor-initramfs-palmetto.testdata.json -> obmc-phosphor-initramfs-palmetto-20180523090537.testdata.json
lrwxrwxrwx 2 aen aen      62 May 23 15:23 zImage -> zImage--4.13.16+git0+468cbec6d2-r0-palmetto-20180523090537.bin
-rw-r--r-- 2 aen aen   24569 May 23 15:23 zImage--4.13.16+git0+468cbec6d2-r0-aspeed-bmc-opp-palmetto-20180523090537.dtb
-rw-r--r-- 2 aen aen 2329848 May 23 15:23 zImage--4.13.16+git0+468cbec6d2-r0-palmetto-20180523090537.bin
lrwxrwxrwx 2 aen aen      77 May 23 15:23 zImage-aspeed-bmc-opp-palmetto.dtb -> zImage--4.13.16+git0+468cbec6d2-r0-aspeed-bmc-opp-palmetto-20180523090537.dtb
lrwxrwxrwx 2 aen aen      62 May 23 15:23 zImage-palmetto.bin -> zImage--4.13.16+git0+468cbec6d2-r0-palmetto-20180523090537.bin
aen@aen-VirtualBox:~/openbmc/build/tmp/deploy/images/palmetto$ 
```

anoo1, 2018-05-31:
> I deleted the openbmc directory then git clone-ed again.
> Is there a make clean equivalent in this yocto build ?

There's `bitbake -c cleanall obmc-phosphor-image`. Also the `build/tmp/` directory can safely be removed.

Aeneas32, 2018-05-31:
In the Term log, can you see any indication why the bitbake does
not complete/succeed ?
Could it be that this build is not compatible with either Virtualbox or Ubuntu 16.04.4 ?
How would you describe this error and from which recipe file is this instigated ?

"ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Function failed: do_generate_static (log file is located at /home/aen/openbmc/build/tmp/work/palmetto-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.10993) "

mdmillerii, 2018-05-31:
That error message gives a log file, and you have to read the log file to see if you can figure out the cause of the error.  The default behavior of bitbake is to grab a few lines from the end of the file which had the file not found error.

That being said, it sounds like the image generation recipe might be missing the dependency on generating the u-boot image as 
1. I don't see any u-boot artifacts ands
2. there was another email [on the list](https://lists.ozlabs.org/pipermail/openbmc/2018-June/011916.html) that appeared to have a similar problem building palmetto.

 @geissonator @bradbishop @amboar

amboar, 2018-06-04:
I haven't had the time to reproduce this, but someone should look into it :)

Aeneas32, 2018-06-04:
Tried to build with /Ubuntu 16.04.4/Virtualbox/3.5GBytes memory/ , still  Fails to build.
Error messages must be vague for so many developers to not be able
to recognize what is the cause of the error.

mdmillerii, 2018-06-25:
@Aeneas32 can you try with https://gerrit.openbmc-project.xyz/#/c/11260/  ?

Aeneas32, 2018-06-25:
Are you saying that you have tested builds with Virtualbox and believe that
downloading the archive from that URL page and attempting the build,
that the build should run to completion ?

mdmillerii, 2018-06-26:
> Are you saying that you have tested builds with Virtualbox

No, I'm saying I noticed the lack of `u-boot` build products in your directory listing, and that missing product was the immediate cause of the packaging step error, and that patch, when applied to the build tree, adds a missing space which was causing that build product to not be produced as a prerequisite to the build step, and therefore I am asking you to test with the proposed fix to the other issue as you have the environment setup.  In other words, I think that is likely to also be the cause of the first issue you have encountered.

See this line from your original post:
```
| dd: failed to open '/home/aen/openbmc/build/tmp/deploy/images/palmetto/u-boot.bin': No such file or directory
```

amboar, 2018-06-27:
> Is this patch incorporated into the current official Openbmc download,

Not yet. We're hoping someone suffering the issue will test it :)

> or if not, what is the procedure to apply this patch to a fresh download of openbmc ?

The procedure is to grab the patch out of Gerrit. There's a download button on the top right which allows you to take several approaches, however I recommend you do a checkout. Run the following in your git tree:

```
git fetch https://gerrit.openbmc-project.xyz/openbmc/openbmc refs/changes/60/11260/2 && git checkout FETCH_HEAD
```

mdmillerii, 2018-08-06:
@Aeneas32 please test with https://github.com/openbmc/meta-phosphor/commit/68fb2e430c53f38d63db4583678559692749190b or later.

gtmills, 2018-08-20:
@Aeneas32 Could you clone the latest openbmc/openbmc and try building?