openbmc/openbmc #3209: OP920:Witherspoon: Medium Ciphers  (>64 bit and <112 bits ) are supported on OpenBMC

cdeadmin, 2018-05-29:


geissonator, 2018-05-29:
Summary from internal defect:
```
Here is the list of medium strength SSL ciphers supported by the remote server :

  Medium Strength Ciphers (> 64-bit and < 112-bit key, or 3DES)

    ECDHE-RSA-DES-CBC3-SHA       Kx=ECDH        Au=RSA      Enc=3DES-CBC(168)        Mac=SHA1
    DES-CBC3-SHA                 Kx=RSA         Au=RSA      Enc=3DES-CBC(168)        Mac=SHA1

The fields above are :

  {OpenSSL ciphername}
  Kx={key exchange}
  Au={authentication}
  Enc={symmetric encryption method}
  Mac={message authentication code}
  {export flag}

As per NIST Standard , ciphers lesser than 112 bits should not be supported
https://csrc.nist.gov/publications/detail/sp/800-131a/archive/2011-01-13
Pls plan to fix it
```

joseph-reynolds, 2018-05-30:
We may just want to wait for and encourage upstream projects to tighten security.
I note that downstream projects (forks, installers, administrators) can change OpenBMC's /etc/sshd settings to satisfy their needs.  In my opinion, that's a half-baked feature that we should mention in our security docs.  I haven't looked at bmcweb's support for changing cipher suites, but it sounds like a step toward to supporting security configurations.

geissonator, 2018-05-30:
> First off, didn't we have a discussion about not including internal tracker numbers (ie SW430293) in public forums/commit messages? I thought the conclusion was that internal defect numbers are for internal use only, and shouldn't appear publicly.

Yeah, we just got a big release out last week.  We're going to regroup and figure out how to get these out of the external github trackers.

> 
> I generally tend to use the mozilla compatibility, as we need to support things that don't necessarily use the best practices (like curl and wget from older OSes) but still provide reasonable security.
> https://wiki.mozilla.org/Security/Server_Side_TLS

Yeah, that looks like it would fix this bug.

> 
> One of the primary designs in bmcweb is the fact that the cipher suites can be software defined, rather than in a fixed config file. The hope was that (hopefully soon) we could make cipher suites configurable at runtime through redfish, so we wouldn't have to field these kinds of "bugs" as often. On another codebase we field this request monthly, and just about every time it amounts to "X specification/security review/researcher says that Cipher suite Y shouldn't be enabled". From there it devolves into a mess of "what compatibility do we need to break to make this change?"
> 

In general though, we still have to have some hard coded default?  Seems like the goal with that should be to be somewhere close to super-secure, but not all the way there.  The mozilla one looks good to me for this.

> In general we try to keep on top of it, but it's error prone. The hope was that at some point, we would just make it configurable with a default from somewhere explicitly defined (like mozilla or OWASP). Those that need backward compatibility, can leave the less secure ciphers enabled. Those that want the latest in security can disable them with a quick command. The defaults will be a compromise.
> 
> In short, if we hardcode this list, it will likely be a losing battle long term. I think we need to look into making it more dynamic.

The dynamic capability sounds like a good goal to me and a nice feature for our users.

rfrandse, 2018-05-31:
<https://gerrit.openbmc-project.xyz/10941> Move nginx ciphers to mozilla guidelines
Resolves  #3209 Witherspoon: Medium Ciphers  (>64 bit and <112 bits ) are supported on OpenBMC

closed 2018-06-04 by geissonator