openbmc/openbmc #3213: sdbusplus bump not working - phosphor-dbus-monitor throwing sdbusplus::exception::SdBusError'

geissonator, 2018-05-30:
Here's the bump: https://gerrit.openbmc-project.xyz/#/c/10878/

Quite a few commits went into it from https://github.com/openbmc/sdbusplus/commits/master

This was a good one, required manual intervention to fix the witherspoon systems it got on.  Here's a bit of the journal:

```
May 30 01:10:55 witherspoon systemd[1]: phosphor-dbus-monitor.service: Main process exited, code=dumped, status=6/ABRT
May 30 01:10:55 witherspoon systemd[1]: phosphor-dbus-monitor.service: Unit entered failed state.
May 30 01:10:55 witherspoon phosphor-mapper[997]: xyz.openbmc_project.Events discovery failure on / - retry
May 30 01:10:55 witherspoon systemd[1]: phosphor-dbus-monitor.service: Failed with result 'core-dump'.
May 30 01:10:56 witherspoon systemd[1]: phosphor-dbus-monitor.service: Service hold-off time over, scheduling restart.
May 30 01:10:57 witherspoon systemd[1]: Stopped Phosphor DBus Monitor.
May 30 01:10:57 witherspoon systemd[1]: Started Phosphor DBus Monitor.
May 30 01:10:58 witherspoon phosphor-dbus-monitor[1232]: terminate called after throwing an instance of 'sdbusplus::exception::SdBusError'
May 30 01:10:58 witherspoon systemd[1]: Started Process Core Dump (PID 1233/UID 0).
May 30 01:11:00 witherspoon systemd[1]: phosphor-dbus-monitor.service: Main process exited, code=dumped, status=6/ABRT
May 30 01:11:01 witherspoon phosphor-mapper[997]: xyz.openbmc_project.Events discovery failure on / - retry
May 30 01:11:01 witherspoon systemd[1]: phosphor-dbus-monitor.service: Unit entered failed state.
May 30 01:11:01 witherspoon systemd[1]: phosphor-dbus-monitor.service: Failed with result 'core-dump'.
May 30 01:11:02 witherspoon systemd[1]: phosphor-dbus-monitor.service: Service hold-off time over, scheduling restart.
```

My best guess is https://github.com/openbmc/sdbusplus/commit/20db3bf2f7aabea70aacfae341a8f43cf5e0dacc so starting this bug with @wak-google

mine260309, 2018-05-30:
Yup, https://github.com/openbmc/sdbusplus/commit/8e20cfbd745db524fdc2ee9f1ee13b3852313f6d checks all the `sd_bus_message_xxx()` calls and throws when it fails.

Probably get a coredump and see which error it comes from, and see if it is the application's issue, or sdbusplus' issue.

wak-google, 2018-05-30:
I was hoping for openbmc/sdbusplus@20db3bf2f7aabea70aacfae341a8f43cf5e0dacc to be integrated into openbmc before openbmc/sdbusplus@8e20cfbd745db524fdc2ee9f1ee13b3852313f6d got integrated. Unfortunately both got merged so this process is going to be manual.

mine260309, 2018-05-31:
I see similar error on `phosphor-chassis-state-manager` as well:
```
        "__CURSOR" : "s=47551038a23b4700932c7f871ea63e0e;i=28d;b=98b3e18f5ede4e5788e593f8538a3ace;m=45adc5c;t=56d7a52ae3456;x=65e381441d59cae1",
        "__REALTIME_TIMESTAMP" : "1527747024139350",
        "__MONOTONIC_TIMESTAMP" : "73063516",
        "_BOOT_ID" : "98b3e18f5ede4e5788e593f8538a3ace",
        "PRIORITY" : "6",
        "_MACHINE_ID" : "06fbe2a362a64a2e8e1261383716a526",
        "_HOSTNAME" : "romulus",
        "SYSLOG_FACILITY" : "3",
        "_UID" : "0",
        "_GID" : "0",
        "_SYSTEMD_SLICE" : "system.slice",
        "_CAP_EFFECTIVE" : "3fffffffff",
        "_TRANSPORT" : "stdout",
        "SYSLOG_IDENTIFIER" : "phosphor-chassis-state-manager",
        "MESSAGE" : "terminate called after throwing an instance of 'sdbusplus::exception::SdBusError'",
        "_PID" : "1007",
        "_COMM" : "phosphor-chassi",
        "_EXE" : "/usr/sbin/phosphor-chassis-state-manager",
        "_CMDLINE" : "phosphor-chassis-state-manager",
        "_SYSTEMD_CGROUP" : "/system.slice/xyz.openbmc_project.State.Chassis.service",
        "_SYSTEMD_UNIT" : "xyz.openbmc_project.State.Chassis.service",
        "_SYSTEMD_INVOCATION_ID" : "75bc8ea31aff49c79d9abf0ee846ee30"
```

wak-google, 2018-05-31:
@mine260309 No, anything doing sdbusplus operations should be ready to catch and handle exceptions for all operations. I'm trying to slowly migrate some of the sdbusplus functions over that have bitten me in the past but almost all of the sd_bus_* calls can return errors and many of them are discarded by sdbusplus. The fixes for all of the current issues were mailed out with the error handling change:
https://gerrit.openbmc-project.xyz/#/c/10609/
https://gerrit.openbmc-project.xyz/#/c/10617/
https://gerrit.openbmc-project.xyz/#/c/10620/

They were supposed to be reviewed / merged before the change which emits the errors. However it got merged early and so these reviews are still in progress. I manually integrated the change to add the exception class to sdbusplus so things should go smoothly once the chassis-state-manager change is integrated.

mine260309, 2018-05-31:
@wak-google Good to know we already have fixes!
Btw, it looks like `phosphor-dbus-monitor`'s fix is missing.

geissonator, 2018-06-01:
@wak-google - were you going to add the fix to phosphor-dbus-monitor?

wak-google, 2018-06-01:
phosphor-dbus-monitor isn't broken by this change, it just doesn't do proper error handling anywhere.

rfrandse, 2018-06-11:
<https://gerrit.openbmc-project.xyz/11018> Add sdbusplus exception handling
Resolves  #3213 sdbusplus bump not working - phosphor-dbus-monitor throwing sdbusplus::exception::SdBusError'

closed 2018-06-12 by geissonator