openbmc/openbmc #3214: BMC shutdown path is taking longer time to reset

gkeishin, 2018-05-30:
The current GA level code takes 30-35 seconds for BMC shutdown path while newer github public takes around 60 seconds consistently.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.2-195"
VERSION_ID="v2.2-195-ga8006e4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.2-195"
BUILD_ID="v2.2"
root@witherspoon:~#
````

```
root@witherspoon:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
```
Issue reboot
```
root@witherspoon:~# reboot
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped target Host and Network Name Lookups.
         Stopping Phosphor Hwmon Poller...
```
From reboot till the U-boot it now takes 1 minutes.. while the older GA level code is well within 30-35 seconds

```
[  253.069756] systemd-shutdown[1]: All filesystems unmounted.
[  253.113897] reboot: Restarting system

U-Boot 2016.07 (May 30 2018 - 02:17:34 +0000)

       Watchdog enabled
DRAM:  496 MiB
```

mine260309, 2018-05-30:
This should be duplicated to https://github.com/openbmc/openbmc/issues/2977

amboar, 2019-03-19:
#2977 has been closed.

closed 2019-03-19 by amboar