openbmc/openbmc #3215: BMC cannot change user password using RESTFUL API

ThaiThu, 2018-05-30:
Hi all,

we want to test restfull api to change "root" user password.
I already logined into the bmc the root account.

And after that, I used below curl command to change the password to ABC:

curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": ABC }'  ttps://${bmc}/xyz/openbmc_project/user/root/action/SetPassword

But the result is:

{
  "data": {
    "description": "Login required"
  },
  "message": "401 Unauthorized",
  "status": "error"
}

Could anyone give me some advises please ?

Thank alot !!!

gtmills, 2018-06-29:
@ThaiThu Still a problem? Can you add the journal? What system?

gtmills, 2018-08-20:
Closing. Need more information. Please reopen if still an issue. Thanks!

closed 2018-08-20 by gtmills