openbmc/openbmc #3219: Investigations in the user management area

dkodihal, 2018-05-31:
Start with reviewing some code that's already out on Gerrit :

https://gerrit.openbmc-project.xyz/#/c/8440/ (Design Doc)

https://gerrit.openbmc-project.xyz/#/c/10113/
https://gerrit.openbmc-project.xyz/#/c/10112/

Goal of the story is to review the code and provide feedback, figure out downstream requirements and how this code can be consumed, come up with emails to the mailing list/telco discussions

dkodihal, 2018-07-12:
Update https://github.com/openbmc/docs/blob/master/user_management.md with a requirements section, to capture requirements on various components - such as the user manager, IPMI, rest server, etc.

ratagupt, 2018-12-18:
We can close this issue as the documents have been updated.

closed 2019-01-24 by rfrandse