openbmc/openbmc #3220: Redfish : Investigate bmcweb

dkodihal, 2018-05-31:
Investigate https://github.com/openbmc/bmcweb. Some of the tasks :

- Try out the webserver via a redfish client.
- Check if redfish backend can be plugged out and plugged to another webserver.
- Investigate how redfish resources are mapped to D-Bus.

Results can be emails to the community, etc.

geissonator, 2018-07-19:
Some initial code to get bmcweb into witherspoon - https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/11509/

jinneal, 2018-09-04:
Could it be in use for AST2500? I try to install the redfish into the AST2500 and access it by https

geissonator, 2018-09-21:
We now have bmcweb in the IBM layer and it has the basic redfish functions working. This story should track more specific work, like maybe helping with https://gerrit.openbmc-project.xyz/#/c/openbmc/bmcweb/+/12018/ and other required features for Redfish.

gtmills, 2019-07-09:
Closing. 
bmcweb is the default webserver.  It includes support for redfish. 
more information at https://github.com/openbmc/bmcweb

closed 2019-07-09 by gtmills