openbmc/openbmc #3221: Set Power Cap from GUI.

gtmills, 2018-05-31:
Currently, the user can only view the power cap not set/enable it. This story is to allow the user to set/enable it.
[image]

gtmills, 2018-06-20:
After talking with Karl, new page under "Server control". Named like "Power consumption".

gtmills, 2018-06-21:
Start of the page 
[image]

[image]

rfrandse, 2018-06-27:
<https://gerrit.openbmc-project.xyz/11291> Set the power cap
Resolves  #3221 Set Power Cap from GUI.

gtmills, 2018-07-06:
[image]

gtmills, 2018-07-09:
Look at having a link from the overview page.

closed 2018-08-24 by geissonator