openbmc/openbmc #3222: GUI: Add functionality to manage BMC and Host time

gtmills, 2018-05-31:
https://github.com/openbmc/phosphor-time-manager

Will need things like TimeSyncMethod (NTP vs Manual), TimeOwner (BMC and/or Host), setting the time

gtmills, 2018-05-31:
Need to work with the design team to come up a new page to do this.

alissaylee, 2018-08-29:
[image]
[image]

- First prompt: User chooses to set date and time a) manually or b) via the NTP server.
- Second prompt: User chooses to set BMC and server's date and time as the same or different.
- Date and Time fields are selectable (and utilize default browser settings). 
- Time zone will be read-only.

alissaylee, 2018-08-29:
[image]
[image]
[image]

**Setting date and time via the NTP server**
- Date, Time, and Timezone are read-only. 
- Initial setup: Text field to set up primary NTP server option is made available.
- Ability to add more NTP server(s).
- Remove NTP server(s) starting with the 2nd NTP server.
- Can "Test" connection before "Saving settings".
- States successful connection of an NTP server with a success icon after user selects "Test" or "Save settings".
- States "Invalid format" text and warning icon for an invalid NTP server entry after user is done typing.
- States "Does not exist or connection problem" text and warning icon for a 'bad' NTP server after selecting "Test" or "Save settings".
- Alerts when user attempts to leave page without saving settings.

alissaylee, 2018-08-29:
To view additional instances for Date and Time Settings, please go here: https://ibm.ent.box.com/folder/52973968475

alissaylee, 2018-09-05:
Public InVision link: https://ibm.invisionapp.com/share/YSNTBXAJGUC#/318326214_Manual_Default_Same

closed 2019-05-27 by stale[bot]