openbmc/openbmc #3223: GUI: Add SNMP settings

gtmills, 2018-05-31:
Need to work with the design team to come up with an enhanced network settings page that includes SNMP or come up with a new SNMP page. 
https://github.com/openbmc/phosphor-snmp is the SNMP repo.

gtmills, 2018-09-04:
Currently, the only SNMP settings available are viewing/creating/deleting SNMP managers. 
https://github.com/openbmc/phosphor-snmp/blob/master/docs/snmp-configuration.md

gtmills, 2018-09-04:
New tab I assume under configuration? "SNMP settings" as the tab name and title of the page?

alissaylee, 2018-09-17:
https://ibm.invisionapp.com/share/YSNTBXAJGUC#/318852407_SNMP_Default

**SNMP settings**
- Initial setup: button to "add snmpv3 manager".
- Fields will include "SNMPv3 manager server" + "port" fields.
- If either field is left empty while the other filled in, field validation will warn user. 
- If both fields are left empty and user selects to save settings, fields will no longer be available on the page.
- Can test manager server connection.
- Can remove a manager field (with both manager server and port).
- Will receive message to save settings if user attempts to leave page without saving edits to the page.
- Field validation for manager server and port fields if there's a connection problem or port's value isn't between 0 to 65,535.

gtmills, 2018-11-06:
Duplicate issue here: https://github.com/openbmc/phosphor-webui/issues/37 Closing.

closed 2018-11-06 by gtmills