openbmc/openbmc #3224: Move to python3

gtmills, 2018-05-31:
https://github.com/openbmc/openbmc/issues/2920 added support for python3

```
 geissonator commented on Mar 4

I think @bradbishop ended up finding that we don't in fact have to update to python 3 within our image. But at some point we'll want to, in order to take advantage of some of the new features in python 3. But not required for yocto 2.4.
```

After moving to python3 should remove python2 
https://github.com/openbmc/openbmc/issues/2994

gtmills, 2018-05-31:
Currently we have 
```
root@Witherspoon:~# python -V
Python 2.7.13

```

gtmills, 2018-05-31:
We can close https://github.com/openbmc/openbmc/issues/1556 when we move to python 3

shenki, 2018-06-01:
Good idea: https://github.com/openbmc/meta-phosphor/issues/13#issuecomment-141889784 :)

gtmills, 2019-03-28:
We removed Python from the image. Closing.

closed 2019-03-28 by gtmills