openbmc/openbmc #3227: Create an IP interface at /xyz/openbmc_project/network/<interface> should return the interface Id.

gtmills, 2018-06-01:
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  POST -d '{"data": ["xyz.oenbmc_project.Network.IP.Protocol.IPv4","10.6.6.7",24,"10.6.6.1"] }' https://${bmc}/xyz/openbmc_project/network/eth0/action/IP
{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}
```
It should return the interface id (e.g. `5c083707`) or the path `/xyz/openbmc_project/network/eth0/ipv4/5c083707`

gtmills, 2018-06-01:
From the GUI has to make another REST call to get the path. No way to calculate beforehand since std::hash is used. https://github.com/openbmc/phosphor-networkd/blob/master/ethernet_interface.cpp#L270

amboar, 2018-06-04:
Do we need an education session on not exposing std::hash() in APIs?

closed 2019-04-27 by stale[bot]