openbmc/openbmc #3232: Can't get the bmc time on evb-ast2500 by rest API

LindarLiu0724, 2018-06-05:
Hi all,
I have built the evb-ast2500 and can't get the bmc time from REST API, could someone give suggestion how to make it works. 
Seems phosphor-time-manager didn't work.

root@evb-ast2500:~# cat /etc/os-release                            
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.2-242"
VERSION_ID="v2.2-242-gec5fb50"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.2-242"
BUILD_ID="v2.2"

and no /xyz/openbmc_project/time/bmc found,
root@openbmc-X8DTL:~# curl -c cjar -b cjar -k -H "Content-Type: application/jso
" https://${bmc}/xyz/openbmc_project/time/
{
  "data": [
    "/xyz/openbmc_project/time/owner", 
    "/xyz/openbmc_project/time/sync_method"
  ], 
  "message": "200 OK", 
  "status": "ok"

LindarLiu0724, 2018-06-05:
How to check?
when run phosphor-timemanager error as below:
root@evb-ast2500:~# phosphor-timemanager 
terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Time::Internal::Error::MethodError'
  what():  xyz.openbmc_project.Time.Internal.Error.MethodError: Failed to invoke Dbus method
Aborted (core dumped)

mine260309, 2018-06-05:
OK, it is built but get run time error.
Please check pretty json journal log to find which dbus method it is failing:
```
journalctl -o json-pretty -t phosphor-timemanager
```

LindarLiu0724, 2018-06-05:
journalctl -o json-pretty -t phosphor-timemanager result:

{
        "__CURSOR" : "s=fad46276973c46b4b14a3b55ef696491;i=27a;b=67a90cc2d6aa45f4bd64748af0190181;m=cd986eb;t=56d63f87cfc1a;x=2171e5d1cf4e62f4",
        "__REALTIME_TIMESTAMP" : "1527651021683738",
        "__MONOTONIC_TIMESTAMP" : "215582443",
        "_BOOT_ID" : "67a90cc2d6aa45f4bd64748af0190181",
        "_UID" : "0",
        "_GID" : "0",
        "_MACHINE_ID" : "08e00565f15c4eb0891aed3fddd260e6",
        "_HOSTNAME" : "jbof-ast2500",
        "PRIORITY" : "3",
        "_TRANSPORT" : "journal",
        "_CAP_EFFECTIVE" : "3fffffffff",
        "CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
        "CODE_LINE" : "100",
        "CODE_FUNC" : "helper_log",
        "_SYSTEMD_CGROUP" : "/system.slice/system-dropbear.slice/dropbear@0-192.168.81.64:22-192.168.81.42:4016.service",
        "_SYSTEMD_UNIT" : "dropbear@0-192.168.81.64:22-192.168.81.42:4016.service",
        "_SYSTEMD_SLICE" : "system-dropbear.slice",
        "_SYSTEMD_INVOCATION_ID" : "10732b7915c14a5486c154cbed4c44bd",
        "MESSAGE" : "Failed to invoke Dbus method",
        "TRANSACTION_ID" : "3152053792",
        "METHOD_NAME" : "GetObject",
        "PATH" : "",
        "INTERFACE" : "xyz.openbmc_project.State.Host",
        "MISC" : "Error reading mapper response",
        "SYSLOG_IDENTIFIER" : "phosphor-timemanager",
        "_PID" : "1092",
        "_COMM" : "phosphor-timema",
        "_EXE" : "/usr/sbin/phosphor-timemanager",
        "_CMDLINE" : "phosphor-timemanager",
        "_SOURCE_REALTIME_TIMESTAMP" : "1527651021683662"
}

mine260309, 2018-06-05:
When phosphor-time-manager starts, it needs to check host state from phosphor-host-state-manager.
And the above log shows that it does not find host state manager's object, and thus it fails to get the service.
I suggest you post the full journal log to see why host state manager is missing.

Btw, please put your code/log in backtick ```, so that it renders as `code/log`.

LindarLiu0724, 2018-06-05:
Thanks your remind:).
I run phosphor-host-state-manager sucessfully by mannual, then can get the bmc time now.
There is no host in our platform, how to handle such issue? 
```
root@jbof-ast2500:~# phosphor-host-state-manager 
```
```
root@openbmc-X8DTL:/home/openbmc/workspace/git/openbmc.git/patch# curl -c cjar -b cjar -k -H "Content-Type: application/json" https://${bmc}/xyz/openbmc_project/time/
{
  "data": [
    "/xyz/openbmc_project/time/owner", 
    "/xyz/openbmc_project/time/bmc", 
    "/xyz/openbmc_project/time/host", 
    "/xyz/openbmc_project/time/sync_method"
  ], 
  "message": "200 OK", 
  "status": "ok"

```

closed 2018-06-26 by LindarLiu0724