openbmc/openbmc #3240: Should allow empty list for type array

gtmills, 2018-06-07:
``` 
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data": [] }' https://${bmc}/xyz/openbmc_project/network/eth0/attr/Nameservers
{
  "data": {
    "description": "Unable to guess signature from an empty list"
  }, 
  "message": "400 Bad Request", 
  "status": "error"
}
```
https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Network/EthernetInterface.interface.yaml#L30

bradbishop, 2018-09-07:
In theory you could introspect the thing you are trying to put, to get the signature first.  It isn't trivial but it shouldn't be too hard either.  Then you could do something like:

```l = dbus.Array([], signature='s')```

for an array of strings for example.

gtmills, 2019-10-02:
Still a problem and referenced in webui code here: https://github.com/openbmc/phosphor-webui/search?q=3240&unscoped_q=3240 
If we move the webui to redfish in these 2 places, will no longer be a problem

gtmills, 2019-10-14:
Appears to be fixed with bmcweb. 
```
 curl  -k -H "Content-Type: application/json" -X PUT -d '{"data": [] }' https://${bmc}/xyz/openbmc_project/network/eth0/attr/NTPServers
{
  "data": null,
  "message": "200 OK",
  "status": "ok"

```
```

 curl  -k -H "Content-Type: application/json" -X PUT -d '{"data": [] }' https://${bmc}/xyz/openbmc_project/network/eth0/attr/Nameservers
{
  "data": null,
  "message": "200 OK",
  "status": "ok"

```

I'll remove the webui hacks for this.

closed 2019-10-14 by gtmills