openbmc/openbmc #3250: GUI: Login screen popping up while doing power operations

jaypadath, 2018-06-15:
When selecting yes option under "Warm reboot" power operations, Login screen comes into the picture.
Because of that automation and all subsequent test cases are getting stopped.

Observed the issue in below other scenarios:
1. While doing cold reboot from webui
2. When logged into the GUI and traversing several tabs 

Frequency of showing the issue is more in mac os environment

gtmills, 2018-06-19:
@jaypadath https://github.com/openbmc/phosphor-webui/blob/master/app/common/services/apiInterceptor.js#L50 redirects to the login screen if the rest server returns 401 (UNAUTHORIZED). We also limit the number of sessions to 16, with the oldest becoming invalid. https://github.com/openbmc/phosphor-rest-server/blob/master/module/obmc/wsgi/apps/rest_dbus.py#L653

gtmills, 2019-01-23:
@jaypadath Can we close? I believe this was a problem with the Robot framework and logging in every time. @sivassrr @rahulmah Are you still seeing?

gtmills, 2019-01-23:
@sivassrr Did you observe this more than once? Can you give more detail? Was there a long period of time between you coming back to the GUI?

rahulmah, 2019-01-23:
@gunnar: This was happening as we were having maximum REST session between and 1st GUI operation is getting killed. Due to which GUI was logging out(we had discuss regarding it some time back over slack).  We have fixed this problem in our automation and now it is working fine. This issue can be closed.

rahulmah, 2019-01-23:
@sivassrr : We should open a separate issue as this was logged because of automation and we have figured it out as commented above.

sivassrr, 2019-01-23:
As we are going to move to Redfish API, Let go ahead and close this. If we see it there then we can open one. In other internal defect tracking, had updated there that it occurs in manual as well. In fact today also experience this.

closed 2019-01-23 by sivassrr