openbmc/openbmc #3252: Remove /dev/mem from kernel configuration

shenki, 2018-06-16:
The presence of this is a security concern. It should be removed from the default kernel configuration.

This should be done after we move to the 4.17 kernel.

AlexanderAmelkin, 2018-06-19:
I'm against it. It is a security concern only in the current state of openbmc where everyone knows the root password and many actions require root access. IMO, the root access should be reserved to openbmc firmware developers only and all users should use non-privileged accounts. That way they won't have a chance to access `/dev/mem`.

`/dev/mem` is available in all Linux distros that I know and nobody considers it a security concern there.

It's all about user privileges.

AlexanderAmelkin, 2018-06-19:
Besides, we are using `devmem` tool in our scripts and that tool requires `/dev/mem`. Removing the device will just break things for us without explicit profit.

amboar, 2018-06-20:
@AlexanderAmelkin you can always add it back in a machine-specific kernel config snippet if necessary.

I agree that user privilege is also a problem that needs to be solved, but I don't see why we would need to keep `/dev/mem` enabled in general.

AlexanderAmelkin, 2018-06-20:
For instance, because it allows debugging Aspeed register misconfiguration on a running system. Again, access to it must be restricted to developers and vendor tech support tier 3+ personnel only.

amboar, 2018-06-20:
Right, but that's a policy decision that should be made by the OEM or system integrator, not something the open source project should be promoting IMO.

Is there something preventing you from adding the config snippet to enable devmem for your machines?

AlexanderAmelkin, 2018-06-20:
@amboar, looks to me like you're proposing to do one extra useless commit to the open source project that will force at least some vendors do more useless commits locally. What's the point?

amboar, 2019-04-02:
Closing this as "mostly removed" with the patch for `mem.devmem=y` on the commandline.

closed 2019-04-02 by amboar