openbmc/openbmc #3259: Commits to phosphor-logger are slow

dkodihal, 2018-06-19:
@wak-google commented on [Tue Jun 19 2018](https://github.com/openbmc/phosphor-logging/issues/9)

I've noticed a 5s+ delay for sending requests to the logger from ipmid, we need to look into ways for speeding this up because it causes unacceptable delays in the ipmi stack. Maybe prefer asynchronous requests as we don't need to block returning up the stack on error reporting.

---

@wak-google commented on [Tue Jun 19 2018](https://github.com/openbmc/phosphor-logging/issues/9#issuecomment-398233575)

https://gerrit.openbmc-project.xyz/#/c/11150/

dkodihal, 2018-06-19:
At the moment there's a D-Bus call from the logging client lib to the logging server to create an error D-Bus object. I think the contract of the commit<>() API is that it creates a D-Bus object. This D-Bus call can be made asyc, but that breaks this contract and I need to think about the impact.

The increase in time though is happening because of  https://gerrit.openbmc-project.xyz/#/c/9910/. I'll discuss the best solution here with the team.

wak-google, 2018-06-19:
I'm not asking for all callers to be made asynchronous, but calls to commit in the ipmi message processing path should probably be made asynchronous since the success of committing an error log doesn't matter before we send the response.

closed 2019-04-27 by stale[bot]