openbmc/openbmc #3266: xyz.openbmc_project.State.Chassis.service trying to read PGOOD state too early

spinler, 2018-06-20:
On Witherspoon, we started seeing:
```
Jun 20 19:24:30 witherspoon phosphor-chassis-state-manager[1434]: Error in bus call - could not get initial pgood
Jun 20 19:24:30 witherspoon phosphor-chassis-state-manager[1434]: Initial Chassis State will be Off

```  
because that property isn't available yet.  This has always been the case I think, and it was unnoticed until someone recently put in that trace.  It doesn't matter much at standby, but with the host on that state will be wrong until eventually it gets straightened out:
```
Jun 20 19:25:14 witherspoon phosphor-chassis-state-manager[1434]: Received signal that power ON is complete
Jun 20 19:25:14 witherspoon phosphor-chassis-state-manager[1434]: Change to Chassis Power State
```

spinler, 2018-06-20:
Going to fix this by putting a `Before=xyz.openbmc_project.State.Chassis.service` dependency into org.openbmc.control.Power@.service.  Works fine now.

rfrandse, 2018-06-21:
<https://gerrit.openbmc-project.xyz/11205> Add dependency to the org.openbmc power service
Resolves  #3266 xyz.openbmc_project.State.Chassis.service trying to read PGOOD state too early

closed 2018-07-19 by geissonator